sipeed/NanoKVM #181: [FEATURE REQUEST] Add VNC support to nanoKVM

LeonardChrist, 2024-12-04:
I would love to be able to connect to the nanoKVM via a standard VNC client instead of the web server. This could have advantages with non-standard web browsers like on a phone or tablet.

I didn't find a lot of info regarding VNC servers on RISC-V, but I suspect that has to do with the lack of mainstream user interfaces for RISC-V. I guess the nanoKVM could simply™ pass the captured frame-textures onto the VNC, an implementation could be possible.

Thank you for your time.

yrabbit, 2025-02-19:
I use remote rendering - where a powerful video card processes Opengl commands and sends me the result. This is done with VirtualGL (https://www.virtualgl.org/).

It was convenient exactly up to the moment when it was possible to slip your own dynamic library to programs. However, the switch from PrusaSlicer to Flatpack has put an end to such manipulations.

When I saw nanoKVM I was delighted - it captures images directly from HDMI, no library replacement required! What was my disappointment when I realised that there is no VNC server :(

The thing is that if my workstation had acceleration I wouldn't need an external GPU, but it seems that in the browser version nanoKVM requires graphics acceleration and it doesn't show anything :(

``` text
[389536:236859584:0219/150143.175489:ERROR:gl_ozone_egl.cc(23)] GLDisplayEGL::Initialize failed.
[389536:236856896:0219/150143.176528:ERROR:gpu_memory_buffer_support_x11.cc(44)] dri3 extension not supported.
[389536:236859584:0219/150143.177252:ERROR:viz_main_impl.cc(186)] Exiting GPU process due to errors during initialization
[389537:236859584:0219/150143.210866:ERROR:gpu_init.cc(521)] Passthrough is not supported, GL is disabled, ANGLE is
```

If nanoKVM had a VNC server, I wouldn't have any problems! :)

Wildcarde, 2025-11-04:
Another usecase for having direct VNC server support would be that we can add it to an apache guacamole instance as a dedicated connection. This would allow you to add any machine plugged into this thing to a 'single pane of glass' config using: https://guacamole.apache.org/

rix1337, 2026-08-31:
> Try https://github.com/onekvm/onekvm-release-nanokvm
> 
> My custom image with NanoKVM , will open-source soon

I don’t see that project supporting vnc @lowtech-guy 
Please elaborate

lowtech-guy, 2026-09-01:
> > Try https://github.com/onekvm/onekvm-release-nanokvm
> > My custom image with NanoKVM , will open-source soon
> 
> I don’t see that project supporting vnc [@lowtech-guy](https://github.com/lowtech-guy) Please elaborate

-> https://github.com/onekvm/onekvm-release-nanokvm/releases/tag/0.1.0