sipeed/NanoKVM #183: No HID on HP Elitedesk 800 G4 SFF

randomdude, 2024-12-05:
My HP 'Elitedesk 800 G4 SFF' machine isn't seeing any keyboard or mouse input from the BIOS. Keystrokes are ignored and the 'hid' indicator on the LCD shows the 'HID' as inactive.

Once Windows boots, the 'HID' indicator becomes active and HID input works okay. 

During BIOS, `dmesg` is flooded with messages like the following:
```
[ 8659.474696] configfs-gadget gadget: usb_gadget_wakeup
[ 8659.508178] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 8659.508194] configfs-gadget gadget: usb_gadget_wakeup
[ 8659.541836] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 8659.541852] configfs-gadget gadget: usb_gadget_wakeup
[ 8659.575308] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
```

Once I boot windows, I see:

```
[ 8659.846409] configfs-gadget gadget: usb_gadget_wakeup
[ 8680.668242] LT6911--:-- INT!!##
[ 8680.856542] LT6911--:-- INT!!##
[ 8681.668816] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
[ 8682.742676] LT6911--:-- INT!!##
[ 8683.023290] LT6911--:-- INT!!##
[ 8685.948442] dwc2 4340000.usb: new device is high-speed
[ 8685.981637] dwc2 4340000.usb: new address 2
[ 8687.456838] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 8687.456854] configfs-gadget gadget: usb_gadget_wakeup
[ 8687.456937] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 8687.456949] configfs-gadget gadget: usb_gadget_wakeup
[ 8687.466161] dwc2 4340000.usb: bound driver configfs-gadget
[ 8687.633621] dwc2 4340000.usb: new device is high-speed
[ 8687.666798] dwc2 4340000.usb: new address 3
[ 8692.643369] dwc2 4340000.usb: wakeup: signalling skipped: is not allowed by host
[ 8692.643387] configfs-gadget gadget: usb_gadget_wakeup
```

I hope these logs are relevant, I don't know if they are. I'm happy to run any commands you need in order to dump USB traffic or suchlike, just let me know how I can help.

randomdude, 2024-12-05:
Sorry, forgot to say - I'm using the PCI-e NanoKVM, firmware is latest (2.1.2). I tried the 'reset HID' function, using a different cable, and even tried the other USB-C port to make sure that wasn't the issue.

Z2Z-GuGu, 2024-12-05:
Hello, thanks for your issue. Please try running "touch /boot/BIOS && restart" on a web terminal. Some computers require HID devices to support a special "BIOS mode" for their BIOS. In NanoKVM, this mode requires creating "/boot/BIOS" to enable it

randomdude, 2024-12-11:
Hi, thanks for the suggested fix. I tried it, but unfortunately it did not correct the problem - there is no change to the situation. 
Googling around, I found [this](https://www.reddit.com/r/sysadmin/comments/18ejszc/keyboard_and_mouse_do_not_work_in_hp_bios_through/?rdt=39012) - I can't speak to how correct it is, but it kind-of suggests there's something weird going on in these machines' BIOS. 

Let me know if there are any commands I can run to log extra information or if I can help otherwise.

randomdude, 2024-12-11:
Ahhh I'm sorry - I just had to reflash my NanoKVM, re-applied the fix, and it works okay, so I must've made some mistake previously! I can confirm that the 'touch /boot/BIOS' fix corrects my issue.

closed 2024-12-16 by Z2Z-GuGu

MarcelSM24, 2025-01-17:
Hi, I'm having the same exact problem, but I tried the "touch /boot/BIOS && restart" and it's not working for me. I have an HP Elitedesk g3 600 so probably it's not that different. Did you change anything specific in the BIOS to make it work? or just reflash? I already tried that and didn't make any difference. And what USB port are you using? Just to make sure of everything because I already tried all.