sipeed/NanoKVM #184: This is embarrassing (for you)

marshalleq, 2024-12-05:
So yet again, after latest update, I cannot access the very thing that is the core promise of your product.  I don't know how I will fix it this time, because I am off overseas in less than a week.  Clearly you latest update you didn't do proper regression testing, perhaps any testing.  What is the purpose of your product if it can't do what you say it will do.

I give up.

Tell me I'm wrong.

Goodbye until random magical future arrives.

diamondiron, 2024-12-05:
As someone who just spent give or take $300 US dollars I feel the exact same way. I have one unit as good as a brick because of this auto update script. See issue #162

marshalleq, 2024-12-05:
Yeah.  Anyway, just to round out what is happening for me this time, I can't access the web page at all.  It's as though the web server isn't running.  Or perhaps the way to access it has changed.  I can still ping it, but haven't figured out what ssh username password I need is yet.  Unplugging from power does not help it.  The screen on the unit though looks pretty normal.  Should be an easy fix if I can get into it.

diamondiron, 2024-12-05:
Okay, I've made some progress. Take out the SD card, flash it with the image on the GIthub releases page with Etcher. Once it's up and running, give it connection to your network. Your network should not be running any private DNS. Private DNS meaning NextDNS, AdGuard, PiHole, Encrypted DNS Servers, whatever. I do not know why but it just absolutely does not work. I setup a mini travel router and enabled plain non-encrypted google 8.8.8.8 on the router config, made absolutely sure it was configured (rebooted the mini travel router).

Power it on, give it ethernet. It will eventually get you to the IP of the device. Once you get IP access, you should be able to login. There should be no reason you cannot, if it is then it is a network issue. Default web-ui username password is admin/admin. The default SSH root username/password is root/root, running on plain port 22.

Immediately after changing the web UI and SSH password, I downloaded this script from another issue here and ran it, after its done running (it doesn't look like its done running, just once you see activity done give it a few seconds then hit enter and if you drop down into terminal you're good), then reboot. Wait again and everything should be okay.

here is the update python script you can upload it or paste it into a file called update.py then run "python update.py"

`import os
import shutil
import time
import zipfile

import requests

temporary = "/root/nanokvm-cache/"

def mkdir():
    is_exists = os.path.exists(temporary)
    if is_exists:
        shutil.rmtree(temporary)
    os.mkdir(temporary)
    print(f"create temporary directory {temporary}")

def read(file):
    with open(file, "r") as f:
        content = f.read()
        return content.replace("\n", "")

def download_firmware():
    print("download firmware...")

    now = int(time.time())
    url = f"https://cdn.sipeed.com/nanokvm/latest.zip?n={now}"
    print(f"download from {url}")

    response = requests.get(url)

    if response.status_code != 200:
        raise Exception(f"download firmware failed, status: {response.status_code}")

    content_type = response.headers.get("content-type")
    if content_type != "application/zip":
        raise Exception(f"download firmware failed, content_type: {content_type}")

    zip_file = f"{temporary}/latest.zip"
    with open(zip_file, "wb") as f:
        for chunk in response.iter_content(chunk_size=1024):
            f.write(chunk)

    with zipfile.ZipFile(zip_file, "r") as f:
        f.extractall(temporary)

    print("download firmware done")

def download_lib():
    print("download lib...")

    device_key = read("/device_key")

    url = f"https://maixvision.sipeed.com/api/v1/nanokvm/encryption?uid={device_key}"
    headers = {"token": "MaixVision2024"}
    response = requests.get(url, headers=headers, stream=True)

    if response.status_code != 200:
        raise Exception(f"download lib failed, status: {response.status_code}")

    content_type = response.headers.get("content-type")
    if content_type != "application/octet-stream":
        raise Exception(f"download lib failed, content_type: {content_type}")

    lib_file = f"{temporary}/libmaixcam_lib.so"
    with open(lib_file, "wb") as f:
        f.write(response.content)

    lib_dir = f"{temporary}/latest/kvm_system/dl_lib/"
    shutil.copy(lib_file, lib_dir)

    print("download lib done")

def update():
    backup_dir = "/root/old"
    firmware_dir = "/kvmapp"

    if os.path.exists(backup_dir):
        shutil.rmtree(backup_dir)

    if os.path.exists(firmware_dir):
        shutil.move(firmware_dir, backup_dir)

    shutil.move(f"{temporary}/latest", firmware_dir)

def change_permissions():
    for root, dirs, files in os.walk("/kvmapp"):
        os.chmod(root, 0o755)

        for file in files:
            file_path = os.path.join(root, file)
            os.chmod(file_path, 0o755)

    print("change permissions done")

def main():
    try:
        print("stop service...")
        os.system("/etc/init.d/S95nanokvm stop")

        print("start update......")

        mkdir()
        download_firmware()
        download_lib()
        update()
        change_permissions()

        version = read("/kvmapp/version")
        print(f"update to {version} success.")
        print("restart service\nthe nanokvm will reboot")
    except Exception as e:
        print(f"update failed\n{e}")
    finally:
        shutil.rmtree(temporary)
        os.system("/etc/init.d/S95nanokvm restart")

if __name__ == "__main__":
    main()
`

marshalleq, 2024-12-05:
Ah yes, it was this script that got me going too.  Forgot about that!

marshalleq, 2024-12-19:
I'm just going to close this issue for now, I got myself going by refreshing the system.  I don't know if the next update will break it again, I'm overseas so I won't be trying that anyway.  I now have another issue where the remote keyboard stopped working, but screen is still working.  So I guess I'll log another ticket for that.

closed 2024-12-19 by marshalleq