sipeed/NanoKVM #186: NanoKVM Lite - Nothing Works

andremain, 2024-12-05:
No video. Tried firmware 1.2.0. , 1.2.1 & 1.3.0.

1.3.0 no web interface shows up.

1. 1.2.0 & 1.2.1 work with no video and trying to update from inside the web interface from 1.2.0. , 1.2.1  from 1.0.0
-> 2.1.2 always fails.

Also downloaded a python script named update-nanokvm.py which also fails both when uploading it from the Web UI and pasting it in the terminal using ssh.

This is a brick!

nwgat, 2024-12-06:
try connecting to it via RNDIS
https://wiki.sipeed.com/hardware/en/maixIII/ax-pi/rndis.html
login with root/root then run `ifup eth0`

nwgat, 2024-12-06:
also are you using normal HDMI or are you using DisplayPort to HDMI cable?

x-magic, 2024-12-07:
It hunted me for a moment or two as well. First boot can take forever to complete and if the device is rebooted then it will brick. 

You can try and connect to the serial and check if the device is fully initialised, then reboot and try again.

lemonderon, 2024-12-17:
Same issue. Whenever I plug in HDMI or USB HID I lose internet connectivity. No DHCP is being retrieved with those cables plugged in. I am on the latest 2.1.2 right now. The product was shipped with 2.1.1. If only AUX power and ethernet is plugged in - DHCP is being retrieved / static config works just fine.

gregoreau, 2025-01-14:
Same thing happened to me, things I tried initially:
- Initially flashed the SD card with image version 1.3.0 but couldn't log into the web UI
- Re-flashed the SD card with image version 1.2.1, this let me log into the web UI but I was stuck at application version 1.0.0, updating via check for updates in the web UI kept failing and there was no video output in the web UI to my proxmox box
- Tried running the local python updating script but it failed as well

In the end, the issue for me was two fold:
- Connectivity to China was blocked on the local LAN by a network policy
- I didn't wait long enough on the initial boot for it to do the initial setup and update

Moved to a network where connectivity to China was not blocked, flashed image version 1.3.0 to the SD card again, booted it up and left it for 30mins, came back and everything was working fine, web UI worked, video and keyboard input to my proxmox box was working and the application version automatically upgraded itself to 2.1.5. Worth noting that I am using a DP to HDMI adapter to plug into the NanoKVM too and it's working fine.

Hope this helps someone, was driving me nuts. I think Sipeed really need a mirror outside of China for the updates, ideally hosting it on github would be nice. That will probably fix the initial setup issue for most people.

The multiple versions are a bit confusing at first as well, if you click on the gear icon at the top of the web UI, it will show you the image version (the one you write to the SD card) and the application version (appears to mean firmware).

haniham, 2025-01-16:
I Tried the following: 

Flashed 1.3.0
Ethernet and Power plugged in with no internet Access

Device hangs in a Loop  (The Adress was served by my test DHCP)
```
Auto-Updating
192.168.177.50
``` 
after some time:
```
Please Connect
to the network
to update app
``` 
It has network, but no Internet
Then it resets and goes back to the updating

I needed to go home so i have Internet access 😒
If i connect my testnet to the internet 
```
Auto-Updating
192.168.177.50
``` 
after some time it boots to the usual Screen with IP and data displayed

So i made a dd dump of the completed Image - was 32GB and zipped it to only 306MB, so just i slight increase from the original Image with 179 MB - why cant we get a complete image. These hardware is not supposed to see any Internet

For owners of  the light Hardware this is more frustrating because you have no idea what the device is doing

Freekers, 2025-02-06:
Same issue here. The only way I could get it to work was as follows:

- Flash 1.2.1
- Wait 5 minutes for the device to boot
- SSH into the device using the super secure credentials root:root
- Run [this (non official) update script](https://github.com/Bengt/NanoKVM-Updater/blob/main/nanokvm-updater.py), because [the official script](https://github.com/sipeed/NanoKVM/issues/42#issuecomment-2339549916) is broken _as well_ (can't make this up).

Last but not least, make sure to not geoblock China in your firewall/network, else it won't work either:  https://github.com/sipeed/NanoKVM/issues/289 (only needed during setup)

The design is very human I must say.

Zaki-XL, 2025-08-24:
A download error occurs because the certificate for "https://maixvision.sipeed.com" expired on Sunday, August 24, 2025 at 8:59:59.

```
2025-08-24 14:26:03,774 - ERROR - Library download failed: HTTPSConnectionPool(host='maixvision.sipeed.com', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?uid=d1bd898c1359c262 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1006)')))
2025-08-24 14:26:03,776 - ERROR - Update failed: HTTPSConnectionPool(host='maixvision.sipeed.com', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?uid=d1bd898c1359c262 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1006)')))
2025-08-24 14:26:03,909 - ERROR - Update process failed: HTTPSConnectionPool(host='maixvision.sipeed.com', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?uid=d1bd898c1359c262 (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1006)')))
```