sipeed/NanoKVM #187: Option to disable Virtual Storage permanently

lord-emperor, 2024-12-05:
Good afternoon.

Please add an option to permanently disable the Virtual Storage feature or advise how I could do this in the firmware. I would like to completely remove the switch from the web UI if possible.

x-magic, 2024-12-07:
I noticed when a 4GB or smaller SD card is being used, the virtual storage partition may not be created, thus renders the function unusable. Could be a workaround.

Z2Z-GuGu, 2024-12-16:
@lord-emperor Hi, would you be open to modifying the configuration through the web terminal instead of making direct changes in the UI to achieve this?

lord-emperor, 2024-12-17:
@x-magic Thanks, that's kind of a great work around. I'll see if I can find a 4GB SD card still up for sale anywhere.

@Z2Z-GuGu Yes of course. Lay the directions on me if you got em.

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

USB settings let an administrator disable the virtual storage function and apply a persistent USB composition. The device removes the storage function from the active USB configuration, and the disabled setting is retained for subsequent boots.

An administrator can still re-enable storage through the settings. This is not irreversible removal of the UI control and does not erase stored ISO files.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/menu/settings/usb/index.tsx) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/service/vm/usb-composition.go).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!