sipeed/NanoKVM #188: [FEATURE REQUEST] XH-HK4401 switch support over serial

azbesthu, 2024-12-06:
There are cheap 4 port hdmi/usb2 switches available on the market. They are not supporting hotkeys, but have a dedicated port for switching inputs. As far as I understand the control port on them is a serial port disguise as a micro-usb connection.

Are there any scripting features for NanoKVM WebUI for placing UI buttons that trigger commands or scripts or gpio / serial communication?

I found this approach on a more pricer other kvm project, with more details about the hardware and serial commands:
https://github.com/pikvm/pikvm/issues/1306
Using with direct UART connection

Z2Z-GuGu, 2024-12-16:
Hello, NanoKVM has reserved two sets of serial ports: "/dev/ttyS1" and "/dev/ttyS2", you can achieve this using the script tool on the webpage along with the built-in picocom.

zarvai-fea, 2025-01-01:
I got my NanoKVM lite and also the HK4401 switch. The switch uses inverted serial so it seems I need to add an inverter chip (74HC14) between them. I guess LicheeRV Nano does not support inverting uart io in a setting by software without the inverter chip.
I will post update with results.

azbesthu, 2025-01-02:
I got my NanoKVM lite and also the HK4401 switch. The switch uses inverted serial so it seems I need to add an inverter chip (74HC14) between them. I guess LicheeRV Nano does not support inverting uart io in a setting by software without the inverter chip.
I will post update with results.

azbesthu, 2025-01-06:
It is working with TX, RX and Ground cable to UART1 :)

HK4401 switch is cutting power on its usb port while changing input, so it needs Y cable for data and power fort NanoKVM lite.
Simple python scripts can be uploaded for every button separately.
[image]

```
import serial

# Open the serial port /dev/ttyS1
ser = serial.Serial('/dev/ttyS1', 19200)  # Adjust baudrate as needed

# Send the string 'G04gA' for button 4
ser.write(b'G04gA')

# Close the serial connection
ser.close()
```

azbesthu, 2025-01-06:
Working fine

closed 2025-01-06 by azbesthu

tomiko23lol, 2025-04-06:
Hello @azbesthu Thanks for info about HK4401 KVM switch and its compatibility with NanoKVM. I would like to try it with serial console controll you described.
I already have this switch and was using it with PiKVM v2 based on Pi Zero 2 W, but I had problem with backpowering of PI. 
When I rebooted PI ( physically or software ) while there was USB or HDMI connected to host machine, Pi failed to boot.  Pi never started this way, so I had to always unplug all hosts, reboot and after sucessfull boot, connect back all host machines. This in the end caused complete failure of my PI. By backpowering of Pi trough CSI bridge from KVMswitch, it killed my PI. 
So my question is ... 

??? Is there any problem with backpowering of NanoKVM from KVM Swtich HK4401, I was thinking about just swapping dead PiKVM to new NanoKVM-Ligh and set up 4 GPIOs as ATX control (power button control) so that I can start or restart each machine by one NanoKVM ???

azbesthu, 2025-04-06:
H @tomiko23lol

I have the NanoKVM lite version, that uses the same usb port for powering the NanoKVM and for dataport to HK4401.
The HK4401 shuts down power when it switches between devices. 
At least my version does that. There are multiple versions of the HK4401, older one may need different datarate setting and also the older version has inverted serial communication. 

So i separated power and data with a little  plitter. 
"USB Power Splitter for Raspberry Pi BliKVM Development Board" . I got one from aliexpress for less than $5.  [link](https://www.aliexpress.com/item/1005007723101153.html)
Also there is a model for 3d printing for this splitter. [link](https://www.printables.com/model/370776-usb-splitter-case-blicube-blikvm/remixes). I needed to add little padding to the usb ports, so it won't move in the case.

[image]

Sometimes NanoKVM needs a reboot after switching between HK4401 imputs, it seems to be better with newer software versions. In older software the usb data process sometimes crashed. But the NanoKVM was accessible, so worst case a quick reboot did it as workaround.

Some computes may not be compatible (yet?) with the hdmi input. I did not get picture with dell vostro 3500 11th gen version. And some other computers may have issue with the usb data - i had trouble with supremicro x9dr7-tf+ with esxi 7, i was not able to control it over its usb port.