sipeed/NanoKVM #190: No HID on Dell Optiplex 3000 / 7000 SFF / Micro

ucki007, 2024-12-06:
Hi,

I own a bunch of Dell Optiplex 3000 and 7000 (micro / small form factor).

I have issue very similar to #183 but with some differences. I'm opening a new issue as there are some differences and I don't want to disturb the communication in the other thread.

What is equal to the thread mentioned above?
1. When booting to the OS, everything is fine: I can see the screen and use mouse and keyboard.
2. When entering the UEFI (BIOS), there is no mouse or keyboard

What differs from the thread mention?
1. I'm using the NanoKVM Full (not the PCIe version)
2. When entering BIOS, the indicator shows that HID is connected - but it is not usable

The proposed solution
`touch /boot/BIOS && reboot`
only solves half of the problem: after that, mouse interaction is available in UEFI (BIOS), but no keyboard interaction.

As mentioned above, I'm using a nanoKVM Full.:
- Image version 1.3.0
- KVM updated to 2.1.2

Any ideas? What can I provide for further investigation?

larryqiann, 2024-12-17:
Potentially irrelevant (or not, since the 30 and 50 series are just cut down versions of the 70 series) but the BIOS on my Dell OptiPlex 5050 works fine with NanoKVM on 1.3.0 and 2.1.2. I can use the mouse (though it is slightly laggy) and view the BIOS screen just fine, though the computer displays "booting in headless mode" on startup for some reason.

ucki007, 2024-12-17:
Hi @larryqiann,

thanks for your reply... the XX50 Series from Dell ist much older then my 3000 / 7000. (Seems like Dell is defining their generations a little strange ;-) )

https://www.dell.com/support/manuals/de-de/optiplex-7050-sff/optiplex-7050-desktop-sff-om/processor-specifications?guid=guid-8ca53ab2-a85d-42d5-9106-5214220306aa&lang=en-us

vs

https://www.dell.com/de-de/shop/desktops-all-in-ones-pcs/optiplex-7000-micro-form-factor/spd/optiplex-7000-micro

It's 6th & 7th gen Intel vs 12th gen

And as written above: my mouse works in UEFI after adding the empty file "BIOS" to "/boot" - but there is no keyboard support available until the OS is booted. So I can't enter the UEFI during startup (only restart into UEFI is possible) or select other boot devices or select to boot into Windows or Linux (via grub boot menu).

One additional information from my side: tried it on my Dell XPS 15 9510 (11th gen from 2021) - same behaviour there. No keyboard, but mouse is working.

ucki007, 2025-02-19:
Hi all,

as there was a bigger update (FW 1.4.0 & SW 2.2.0) I tried again, including the flash: still no further improvement.

ucki007, 2025-03-26:
As commented in #239 - problem is solved in software version 2.2.5

In HID only mode, keyboard and mouse are working in UEFI on my Optiplex systems.

Thanks a lot!

closed 2025-03-26 by ucki007