sipeed/NanoKVM #191: Correlation of Application version to GitHub releases?

ashleyw-gh, 2024-12-06:
If I look at the NanoKVM (that has been updated from the UI) it's showing that I'm on the latest release the the UI but its showing as version 2.1.2;
[image]

How does version 2.1.2 correlate to any of the GitHub releases?

ashleyw-gh, 2024-12-06:
FYI it appears that image version 1.3 correlates to application version 2.1.2 from a trawl of the issues.

ucki007, 2024-12-06:
Hi,

as far as I understand there is no direct correlation.
The GitHub release found under https://github.com/sipeed/NanoKVM/releases is the image version - at least for me, that is 1.3.0. According to the date in your image version, you should be 1.2.1

I don't understand, where the application comes from - but I assume that is another (currently non public repository?!?)

And no - Image Version 1.3.0 does not correlate to Application Version 2.1.2. I flashed Image Version 1.3.0 a few days ago and it came up with Application Version 2.1.1. The update to 2.1.2 came to my devices today.

Took me a while to understand - the image version can only be updated by flashing the SD card. The Application can be updated online.

ashleyw-gh, 2024-12-06:
ah thanks - that explains it - I pulled down the source locally and then did a string search and could find no relevant reference to "21.2" so as you say there are presumably other closed box components - which is a little strange as it should be possible to build the firmware release from the source - otherwise what is the point of this public GitHub project?
Don't get me wrong the NanoKVM is an awesome piece of kit, and for my simple use case its great and rock solid.

alexisfrjp, 2025-01-23:
> ah thanks - that explains it - I pulled down the source locally and then did a string search and could find no relevant reference to "2.1.2" so as you say there are presumably other closed box [...]

But it defeats the purpose if the app isn't opensource. App that has access to the video, my keyboard inputs and all.

alexisfrjp, 2025-01-23:
Where is the sourcecode of the app?

alexisfrjp, 2025-01-23:
2.x.x the update tab refers to, seems to be the NaniKVM-system.

https://github.com/sipeed/NanoKVM-System