sipeed/NanoKVM #192: Outdated versions of Tailscale are being installed

lwbt, 2024-12-07:
First of all, please note that some users browsing here have stated that they are Tailscale employees and that they can offer help. (No I'm not one of them.)

https://github.com/sipeed/NanoKVM/blob/main/server/service/network/tailscale.go#L71

```go
		downloadUrl = "https://cdn.sipeed.com/nanokvm/resources/tailscale_riscv64.zip"
```

With increasing frustration, I have to ask: Why are you doing this? I'm certain Tailscale has a CDN, probably a better one than yours. If you really have to do it this way, then at least update the asset so that it is not plenty of SemVer versions or months behind.

Filadeus, 2024-12-07:
Currently the latest version of Tailscale struggles to run on nanoKVM, it gets killed by OOM reaper minutes after authenticating the device, so by hosting an older version they probably getting around this issue.

lwbt, 2024-12-07:
Stop the nonsensical reasoning and guessing. If there is a need to use older versions, then let experienced users choose from a menu of versions from the official site. Hosting an outdated version on any other server is just malpractice.

Tailscale got OOM killed because there was not enough memory on the system, there is no problem with the applikcation itself. The 1.3 images posted here in the release section allocates less memory to the imaging system and leaves now 150 MB for user space applications.

mochechan, 2024-12-09:
About tailscale, my nanokvm can work in a short period. 
Tailscale crashes after minutes, and there's no any log.
If the reason is not enough memory, I don't know why my nanokvm only has 128MB RAM.
[The nanokvm should has 256MB RAM.](https://wiki.sipeed.com/hardware/zh/lichee/RV_Nano/1_intro.html)

lwbt, 2024-12-09:
> About tailscale, my nanokvm can work in a short period. Tailscale crashes after minutes,...

This is not a discussion forum, this is an issue tracker. Stop hijacking issues with topics addressed in other issues.

https://github.com/sipeed/NanoKVM/releases/tag/v1.3.0

> 3. Expand user space RAM from 128MB to 158MB

Upgrade your device.

wj-xiao, 2025-02-19:
We've updated Tailscale's download source to the official address.

You can upgrade Tailscale in two ways:

1. [Flash](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) the image to [v1.4.0](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) , where includes the latest Tailscale version.
2. Update the application to [2.2.0](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0) , then uninstall and reinstall Tailscale via the web settings. 

For OOM issues, please refer to #149 .

closed 2025-02-19 by wj-xiao