sipeed/NanoKVM #193: No Webinterface after some time

th3cube, 2024-12-07:
I have a Problem with the NanoKvm Full version 

Image version 1.3.0
App 2.1.2

After some time the web interface is gone and I can only access it trough ssh

nothing is listening on port 80 anymore 

`# ss -tulpn
Netid           State            Recv-Q            Send-Q                                     Local Address:Port                        Peer Address:Port           Process
udp             UNCONN           0                 0                                                0.0.0.0:41641                            0.0.0.0:*               users:(("tailscaled",pid=471,fd=19))
udp             UNCONN           0                 0                                                0.0.0.0:51016                            0.0.0.0:*               users:(("avahi-daemon",pid=408,fd=14))
udp             UNCONN           0                 0                                                0.0.0.0:5353                             0.0.0.0:*               users:(("avahi-daemon",pid=408,fd=12))
udp             UNCONN           0                 0                                                0.0.0.0:1900                             0.0.0.0:*               users:(("ssdpd",pid=414,fd=6))
udp             UNCONN           0                 0                                           10.10.10.240:123                              0.0.0.0:*               users:(("ntpd",pid=402,fd=51))
udp             UNCONN           0                 0                                              127.0.0.1:123                              0.0.0.0:*               users:(("ntpd",pid=402,fd=50))
udp             UNCONN           0                 0                                                0.0.0.0:123                              0.0.0.0:*               users:(("ntpd",pid=402,fd=49))
udp             UNCONN           0                 0                                                      *:41641                                  *:*               users:(("tailscaled",pid=471,fd=18))
udp             UNCONN           0                 0                                                      *:5353                                   *:*               users:(("avahi-daemon",pid=408,fd=13))
udp             UNCONN           0                 0                                                      *:39289                                  *:*               users:(("avahi-daemon",pid=408,fd=15))
udp             UNCONN           0                 0                            [fe80::2ffd:3279:1b19:c627]:123                                    *:*               users:(("ntpd",pid=402,fd=57))
udp             UNCONN           0                 0                            [fe80::4ada:35ff:fe6f:59c9]:123                                    *:*               users:(("ntpd",pid=402,fd=53))
udp             UNCONN           0                 0                                                  [::1]:123                                    *:*               users:(("ntpd",pid=402,fd=52))
udp             UNCONN           0                 0                                                      *:123                                    *:*               users:(("ntpd",pid=402,fd=48))
tcp             LISTEN           0                 0                                                0.0.0.0:1901                             0.0.0.0:*               users:(("ssdpd",pid=414,fd=3))
tcp             LISTEN           0                 0                                                0.0.0.0:22                               0.0.0.0:*               users:(("sshd",pid=419,fd=3))
tcp             LISTEN           0                 0                                                      *:22                                     *:*               users:(("sshd",pid=419,fd=4))`

when I do "/etc/init.d/S95nanokvm restart" it tries to restart but I get something like "out of memory"

from there only a reinstall over the python reinstall script work. Then the web interface comes back for some time only to disappear again. Not sure if this startet with 2.1.2

duffysk, 2024-12-11:
@th3cube could you please share python reinstall script work?

th3cube, 2024-12-15:
@duffysk 
ssh into your nanoKVM
cd /etc/kvm/

and run the update script
./update-nanokvm.py

Z2Z-GuGu, 2024-12-16:
@th3cube Hello, could you please paste a log the next time the system fails to boot? This would help us continue to improve. Additionally, if you receive a memory shortage message, you might be able to resolve it by using reboot.

th3cube, 2024-12-18:
> @th3cube Hello, could you please paste a log the next time the system fails to boot? This would help us continue to improve. Additionally, if you receive a memory shortage message, you might be able to resolve it by using reboot.

Will paste the Logs here the Next Time it Happens. I already did try the Reboot but that didn't  work. I had to use the Python Script to bring it back

daemontus, 2024-12-27:
Hi! I seem to be running into the same issue. Here is the output from `python update-nanokvm.py`, `/etc/init.d/S95nanokvm restart` and from `/var/log/messages`. As in the original post, the problem goes away temporarily if I do the update. Also, as should be clear from the logs, I'm running `tailscale`, so that might be contributing to higher resource usage.

Z2Z-GuGu, 2024-12-30:
> Hi! I seem to be running into the same issue. Here is the output from `python update-nanokvm.py`, `/etc/init.d/S95nanokvm restart` and from `/var/log/messages`. As in the original post, the problem [...]

Hello, thank you for your log. I have analyzed that the error was caused by incorrect parameter passing in the image processing system. Could you please execute echo "$(cat /kvmapp/kvm/width) * $(cat /kvmapp/kvm/height)" when the error occurs and check if it matches the HDMI output resolution on the host?

Additionally, I need to know the version of your hardware, so please execute cat /etc/kvm/hw and cat /etc/kvm/hdmi_version separately. Thank you!

daemontus, 2024-12-30:
Thank you for the reply! At the moment, the device is working. I'll test if/when the failure occurs again. 

One more thing that I noticed that might be relevant: the computer I'm connected to seems to let the display "go to sleep" eventually. At that point, I have to press some key (through the remote interface) to wake up the screen. So far, this has been working ok, but maybe this is related to the failure somehow, because when the display is "sleeping", the width and height values are not correct.

Output when display is sleeping:
```
# cat /kvmapp/kvm/width 
286
# cat /kvmapp/kvm/height 
1
```

Output when display is working:
```
# cat /kvmapp/kvm/height 
1080
# cat /kvmapp/kvm/width 
1920
```

Other HW info:
```
# cat /etc/kvm/hw 
beta
# cat /etc/kvm/hdmi_version
c
```

Z2Z-GuGu, 2025-01-06:
> Output when display is sleeping:

This may be due to not capturing the normal resolution. The standard HDMI resolution is something like '1920*1080', but it seems your device has captured a screen size of '1*286', which prevents the image input system from initializing correctly. If a similar situation occurs again, and if possible, try unplugging and re-plugging the HDMI while the webpage is open. If you're remote, you can manually modify the values of "/kvmapp/kvm/width" and "/kvmapp/kvm/height", and then `reboot` to restart.

Krusenbaum, 2025-01-08:
I have the same issue with a brand new device. Received today, first session went fine, left it open for 20 min and now the web interface is refusing connections. Restarts do not help.

daemontus, 2025-01-08:
@Krusenbaum make sure to run `python update-nanokvm.py` over SSH and then restart (default SSH password is `root`). I did that and so far the device has been working ok for almost a month now :)

Heath3827, 2025-02-09:
I had the issue of the KVM going to eternal sleep when left unattended for a short period of time. I tried running the `update-nanokvm.py` script and all came back ok. Then I left it overnight only to find it had entered another eternal sleep, port 80 was no longer open, and the terminal window was full of the following output:

```
**First part ran to the top of the terminal window and beyond the max lines.**
[2025-02-08 15:50:02] [error] [cgo.go:64] failed to read kvm image: -1
[2025-02-08 15:50:03] [error] [cgo.go:64] failed to read kvm image: -1
maix multi-media driver destroyed.
ISP Vipipe(0) Free pa(0x8cae8000) va(0x0x3fdd98a000)
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 942
pstSize->u32Height = 1732
pstSize->u32Width = 942
pstSize->u32Height = 1732
pstSize->u32Width = 942
pstSize->u32Height = 1732
ISP Vipipe(0) Allocate pa(0x8c608000) va(0x0x3fdd98a000) size(291424)
pstSize->u32Width = 942
pstSize->u32Height = 1732
stSnsrMode.u16Width 942 stSnsrMode.u16Height 1732 60.000000 wdrMode 0 pstSnsObj 0x3fddf652e8
pstSize->u32Width = 942
pstSize->u32Height = 1732
pstSize->u32Width = 942
pstSize->u32Height = 1732
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[lt6911_probe] -208 = exit success 
pstSize->u32Width = 942
pstSize->u32Height = 1732
20250208 15:50:05.220 1065 E isp AE_GetStatisticsConfig:4202 pstAeStatisticsCfg is NULL.
20250208 15:50:05.221 1065 E isp AE_SetStatisticsConfig:4188 pstAeStatisticsCfg is NULL.
20250208 15:50:05.222 1065 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
[... 427 lines cut ...]
NanoKVM-Server/service/stream.GetFrameRateCounter.func1.1()
	/home/ubuntu/code/server/service/stream/frame-rate.go:32 +0x7c fp=0x3f941b8fd8 sp=0x3f941b8f18 pc=0x5ef674
runtime.goexit({})
	/usr/local/go/src/runtime/asm_riscv64.s:540 +0x4 fp=0x3f941b8fd8 sp=0x3f941b8fd8 pc=0x81cb4
created by NanoKVM-Server/service/stream.GetFrameRateCounter.func1 in goroutine 49
	/home/ubuntu/code/server/service/stream/frame-rate.go:28 +0x5c

ra  0x3fdfddbdbe	sp  0x3f98532ce0
gp  0xd4b858	tp  0x3f985332d8
t0  0x1000	t1  0x3fddb00a6c
t2  0x0	s0  0x3f98532ce0
s1  0x0	a0  0x0
a1  0x3f98532ce0	a2  0x0
a3  0x8	a4  0xffffffffffffffff
a5  0xfffffffffffff000	a6  0xa
a7  0x87	s2  0x3fdddf75e0
s3  0x3f98532e90	s4  0x780
s5  0x8	s6  0x1
s7  0x3	s8  0x1
s9  0x780	s10 0x438
s11 0x3fdfd636e8	t3  0x3fdfdaba50
t4  0x10	t5  0xfffffffffffffffd
t6  0x0	pc  0x3fdfddbc30
#
```