sipeed/NanoKVM #194: Just doesn't work.  Not fit for purpose.

MrHC1983, 2024-12-08:
Just got mine. 

1.  None of the USB ports are labeled on the device.  So you have to goto the internet to find a manual or someone who's hooked it up.  Such a basic thing, add some stickers or something.
2.  It when I do have it working performs OK, not as fast as I could like response wise, you have to fiddle with the settings.
3.  99% of the time it just does not work:

To explain.  So you have ATX Control via USB-C to the header adapter etc, that's fine can only go into ONE unlabeled port on the unit, if you plug it in anywhere else it doesn't turn on.  Then you have two other USB's....... one is for PC Input (KB/Mouse) also powers it and also another USB for AUX Power.  Now........  I did a test run on mine, so shut it down the KVM powered off no USB power etc.  So I connected AUX Power, just a standard run of the mill USB-A charger to the USB power AUX port on the device.  It fired up OK.

But when it fires up 99% of the time it does not get a network IP address, I have tried it now with different cables, switches even a different modem to see......... It's faulty.  It just is so inconsistent and this device is meant to be THE CONSISTANT thing to get your system going if it's off. 

I have the latest update so I have no idea what's wrong with it.  There should also be a button to reset the PiKVM itself both in the software and on the front of it.......  

Ashame, I'll be sending mine back as it's not fit for purpose.  A real shame we roll out probably one a month with servers etc (PiKVM's) ........  so I got this to try but it'd a dud.

MrHC1983, 2024-12-08:
1.
Alright so first thing I found............. this unit has a 100Mbps network only port.  So when I had it working I just happened to have it in a 100Mbps port on a device..........  but it won't work on anything auto gigabit or above.  Lucky for me I have it in an Omada based device so I just set the port to 100Mbps and sure enough it gets an IP now.

2.

It still has this weird issue when AUX Power is connected in.........  so a USB-C to PowerPoint USB Charger......... it fails to work properly it's super laggy, pull the power and just have the PC Input power it works fine.  This needs to be fixed, I read somewhere maybe my model is PoE capable so I will next try that and unplug the AUX Power and try that.

MrHC1983, 2024-12-08:
Coming back to part 2............. so it just won't work properly if I have aux power in.............  just pick ANY USB-A charger......... nope.  It's on but the mouse won't work when it unlocks etc.  I managed to just plug the AUX power into my CCTV unit close by as it had spare USB ports, confirmed the PiKVM doesn't go off and just works............ but this needs to be fixed.  

And the FINAL ............  biggest thing is performance.

My mouse sucks on this.........  WHY?  It's so slow, it stutters, it doesn't appear to be the unit in terms of usage.........  please improve this.

patschi, 2024-12-08:
> 1. None of the USB ports are labeled on the device.

They are labeled on the sides of the device:
[image]

> My mouse sucks on this......... WHY? It's so slow, it stutters

Are you using it in the local network? Resolution and FPS reduced? MJPEG or H264? Is the screen busy, or a slowly-changing Linux terminal or similar?

cretudorin, 2024-12-09:
> Alright so first thing I found............. this unit has a 100Mbps network only port. So when I had it working I just happened to have it in a 100Mbps port on a device.......... but it won't work [...]

That might be an issue from your network device, normally should auto negociate down

patschi, 2024-12-09:
> That might be an issue from your network device, normally should auto negociate down

Not all network devices can. See #21. (But I believe other Enterprise KVMs, e.g. built-in in HPE or so, are using 100M...)

cretudorin, 2024-12-09:
Well, in that case nanokvm is not the right device for you because the hardware only supports 100mbit Ethernet

I don't see this as an issue with nanokvm itself

MrHC1983, 2024-12-10:
I know it's labeled on the side of the device but it would have been better to just have it printed above each of the ports themselves.  PC, AUX, ATX etc......

And I really dunno why it has issues when AUX is plugged in.  Mine will not work at all on a USB charger, I've tried a few.......    it stays on but it's unusable.  Perhaps you guys need to modify the design and supply a charger with it that works?

I've noticed when it's set to H.264 it works even tho the mouse and screen is behind you can click around and it will catch up.  I was trying to FORCE the MTU to 1465 maybe 1480 by default through the terminal but couldn't make it stick, everytime it reboots it changes back.  I wanted to change it to help response times.

The MJPEG option is more laggy then the H.264.  I dunno this is your project only you know if there's room to improve the performance.  I would really have loved it if it worked on iPhone.  It cannot do virtually anything from the iPhone so I'd need another computer to be able to work with it.

It's not bad but it's not quite at the PiKVM level I've been rolling out as of late so...

image.png (view on web)<https://github.com/user-attachments/assets/557f1eb2-6371-4fa2-b786-1344281637cf>

lemonderon, 2024-12-17:
@cretudorin I think you might have an issue with grounding as I do #208

ThreepE0, 2024-12-17:
Imagine opening an issue for "it would have been better if..."

dotX12, 2025-02-01:
@MrHC1983 see: https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317

MrHC1983, 2026-10-02:
Well .... I never solved my issue I see these grounding posts.

Mine now has a 404 error when going to the IP.  But as I've had it ages now.  I've got it powered from the USB port on my CCTV box but it really doesn't matter what I power it from eventually the keyboard and mouse stop working.  Either way it's faulty out of the box and not fit for purpose.

I assume the SD card has died or something with my current error not sure.  All I know is everytime I come to use it, it does not work.  

I hoped the updates would fix it but it's just faulty.

And stop making excuses and noting I should have done this or that.

At the end of the day it does not work as intended.

I regret the purchase and have decided it's time to remove it because it's pissing me off and buy another from someone else.

If I am remote I do not have time for this nonsense.  

The way I have always fixed its weird issues pull both the usb's out so it reboots ...

Can't do that remotely which is what it's meant to be used for.