sipeed/NanoKVM #195: HDMI frozen; unable to send keyboard/mouse events to target

sidcha, 2024-12-08:
I had a NanoKVM plugged into a server for many months. Today, I had to access the BIOS page of that server and when I did, the screen just froze (unable to send keyboard/mouse events) in one of the BIOS pages. Unfortunately, I don't have the ATX board plugged into this server so I cannot reboot it.
 
Things I've tried:
- Clicked on the "Reset HID" option on the UI, nothing changes
- Reboot the NanoKVM, no luck
- SSH-ed into the NanoKVM and tried to poke around, but TBH, haven't seriously looked.

It appears my server would just stay in that BIOS page indefinitely and I cannot get back to server (physically) for some time. So I would like some suggestions of what I can try to debug this issue. I consider myself to be a fairly advanced user, and I'm desperate; so please suggest any means to recover/restart my server.

sidcha, 2024-12-11:
I found that the check for updates button in the website does not work anymore (failed to get version). Turns out this is a common problem? I tried the manual udpate script posted elsewhere but that cannot resolve cdn.sipeed.com to fetch the latest binaries. So I ran the update script elsewhere, moved the contents to desired location and got the updater to work but that didn't launch the webserver automatically.

I suspected that the latest release caused some regression and tried to go back to an older release. I can't find any automated means of doing this so I manually moved /root/old to /kvmapp and did `/etc/init.d/S95nanokvm restart`. Again, this did not launch the web server so I had to manually invoke `/kvmapp/server/NanoKVM-Server`. Again I have the same problem as before.

sidcha, 2024-12-11:
Why do you have only the latest version in the update server? What if I want to go back many versions in the past?

An update breaking further udpates is super critical path to test before making a new release. And rollbacks are a real thing. People use.

At this point I think this product is maintained like a high-school project. Serves me right for throwing somethign this alpha to monitor critical servers.

Z2Z-GuGu, 2024-12-16:
I'm very sorry for the poor experience you've had. 
The reason the keyboard and mouse may not be recognized in the BIOS could be that they haven't entered BIOS mode. You can enable it by executing 
```shell
touch /boot/BIOS
reboot
```
Regarding the CDN issue, you can add the following lines to `/boot/resolv.conf`:
```shell
nameserver 119.29.29.29
nameserver 223.5.5.5
```
Then, reboot the system.

sidcha, 2024-12-16:
> The reason the keyboard and mouse may not be recognized in the BIOS could be that they haven't entered BIOS mode

I was already in BIOS, navigating to a page when thigs stopped working mid-way. So it _was_ in BIOS mode already? Anw, I tried creating `/boot/BIOS` as suggested and it didn't help.

What is the purpose of the reset HID button in the UI? Is there a way to manualy trigger this?  Is there anything else I can try?

> Regarding the CDN issue, you can add the following lines to /boot/resolv.conf

I fixed ths CDN problem by looking at other similar threads. I was trying roll-back to the `/root/old` directory maually by reading what the python script was doing and tings are really broken now. The web server does not auto start after boot so I have to execure `/kvmapp/server/NanoKVM-Server` manually; evenso, the update fails with "failed to get version".

For now, I've resorted to just poking around the system as-is, to avoid any further breakage (as I'm not server  adjacent ATM).