sipeed/NanoKVM #196: Reduce Latency

corrpel, 2024-12-08:
Hello,
Any plans to improve the latency?
What i see now is ~ 160ms (h264).
I don’t want to do any gaming but a consistent remote good experience would be 50-100ms.

azsde, 2024-12-09:
I have the exact same experience, I wish the latency was better.

mcury1985, 2024-12-09:
I use KVM just to boot, emergencies like OS install or recover, BIOS settings..
For everything else, I'm just using RDP.

mcury1985, 2024-12-10:
> @mcury1985 that sounds good. Am i using it to remote to a PC that has no access to local network and i am not able to install or enable anything on it, just because the source laptop is worse than my mackbook (mainly the display that doesn’t allow me to stay outside in bright light)

For RDP to work, you would need that PC to have network access.
If it is remote, you could set up a VPN to your router, or run tailscale on it, I don't recommend port forward for this purpose.

The strong side of the KVM for me is the ability to boot the device if it hangs and I'm far away.
Turn it off, boot and recover if necessary.
I leave a few .isos in the data folder and that allows me to run Windows recovery, arch-chroot Arch linux installations and so on.
This saves me to have to drive to the customer premises.