sipeed/NanoKVM #197: [SECURITY] NanoKVM is a router

ValdikSS, 2024-12-09:
NanoKVM [v1.3.0 image](https://github.com/sipeed/NanoKVM/releases/tag/v1.3.0) (20241120_NanoKVM_Rev1_3_0.img.xz) comes with IPv4 and IPv6 routing enabled, with firewall accepting all routed connections unconditionally.

When IP forwarding is enabled, the device can potentially forward packets between different networks, which might expose internal network to unauthorized access or attacks.  
https://www.tenable.com/plugins/nessus/50686

NanoKVM is not a router, hence this function should be disabled.

```
# sysctl -a 2>/dev/null | grep -E 'ip_forward|eth0.*forwarding'
net.ipv4.conf.eth0.bc_forwarding = 0
net.ipv4.conf.eth0.forwarding = 1
net.ipv4.conf.eth0.mc_forwarding = 0
net.ipv4.ip_forward = 1
net.ipv4.ip_forward_update_priority = 1
net.ipv4.ip_forward_use_pmtu = 0
net.ipv6.conf.eth0.forwarding = 1
```

```
# iptables -vnL FORWARD
Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
# ip6tables -vnL FORWARD
Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
```

Forwarding is enabled by `/etc/sysctl.d/99-tailscale.conf`, which I assume is not necessary for the client configuration.

```
# cat /etc/sysctl.d/99-tailscale.conf 
net.ipv4.ip_forward = 1
net.ipv6.conf.all.forwarding = 1
```

mcury1985, 2024-12-09:
Good, thanks for informing.
I have a DMZ, /30, just for the nanoKVM. This network doesn't have IPv6 enabled and it is blocked to reach any other VLAN in my network, only Internet access is allowed (for tailscale).

Ahrotahntee, 2024-12-13:
> Forwarding is enabled by /etc/sysctl.d/99-tailscale.conf, which I assume is not necessary for the client configuration.

Additional information; This may be in preparation for the kvm to run tailscaled with routes advertised. When tailscale is configured without routes advertised this file is blank.

The behavior should be the opposite, it should default to no forwarding and only enable forwarding if routes are being adverised. Tailscale's status json object contains a sub-object `self` which includes a list of routes if it's advertising routes.

AJolly, 2025-01-24:
This should be configurable.  I like having nanokvm setup as a tailscale router

wj-xiao, 2025-02-06:
The `/etc/sysctl.d/99-tailscale.conf` is created by the Tailscale startup script `/etc/init.d/S98tailscaled`.

In the next version, Tailscale will not start automatically at boot. IP forwarding will only be enabled when Tailscale is started manually and disabled when stopped.

Gunni, 2025-02-06:
@wj-xiao note that doing so means enabling tailscale will disable IPv6 unless #297 is also fixed.

markuman, 2025-02-06:
In the meantime, empty the file and make it immutable  
https://github.com/markuman/nanokvm-mitigations/blob/latest/cleanup.yml#L41-L48

Sfinx, 2025-02-07:
This stopped me from buying this china spy shit - https://youtu.be/plJGZQ35Q6I?t=1386

wj-xiao, 2025-02-19:
You can [Flash](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) the Image [v.1.4.0](https://github.com/sipeed/NanoKVM/releases/tag/v1.4.0) , Tailscale is disabled by default. The router is also disabled by default, and the `/etc/sysctl.d/99-tailscale.conf` is created only when Tailscale starts.

Alternatively, update the Application to [2.2.0](https://github.com/sipeed/NanoKVM/releases/tag/2.2.0) , and disable Tailscale in the web settings. This deletes the `/etc/sysctl.d/99-tailscale.conf` and prevents Tailscale from starting on boot.

closed 2025-02-19 by wj-xiao

alexisfrjp, 2025-11-14:
> This stopped me from buying this china spy shit - https://youtu.be/plJGZQ35Q6I?t=1386

nobody cares about your choice, and you're still here commenting...