sipeed/NanoKVM #199: Documentation for scripts

gregology, 2024-12-09:
The NanoKVM is great. However, I can't find any documentation on writing scripts. Is there any example scripts or documentation online?

For context, I have a dual boot Ubuntu/Windows PC. I want to create a script to boot into Windows which would require pressing the power button, waiting some time, pressing the down arrow a few times, and then pressing the enter key.

This is the pseudo code

```
power on
wait 27 seconds
press down arrow 4 times
press enter key
```

I've tried this bash script

```bash
#!/bin/bash

echo "power" > /dev/hidg0
sleep 27

for i in {1..4}; do
    echo "down" > /dev/hidg0
done

echo "enter" > /dev/hidg0
```

and this Python script

```python
import time

def send_key(key):
    with open('/dev/hidg0', 'w') as f:
        f.write(key)

send_key("power")
time.sleep(27)

for _ in range(4):
    send_key("down")

send_key("enter")
```

But both fail when I try to run them. I have also tried sending messages to `/dev/hidg1`, `/dev/hidg2`, and `/dev/hwrng` in the terminal with no success. Any direction would be appreciated. Cheers!

fenio, 2024-12-14:
I don't want to duplicate issues so I'm just commenting this one as I think I could benefit from that documentation too.

I'm using NanoKVM in front of Unitek's HW KVM to be able to have access to several nodes of my homelab.
That HW KVM supports some hotkeys to switch between inputs without touching hardware.
[image]
I guess ones scripts of NanoKVM would be documented and allow to run them with some other hotkeys I could switch between several HW inputs from my laptop. So the question is how to write this NanoKVM scripts and preferrably how to lunch them with shortucts/hotkeys?

Z2Z-GuGu, 2024-12-16:
Perhaps I should provide detailed operational methods in the wiki. For now, I will share the known solutions to the issues here, and everyone is welcome to @ me with questions.
+ Power button operation:
``` shell
echo 1 > /sys/class/gpio/gpio503/value # Key pressed
sleep 1                                                   # Delay
echo 0 > /sys/class/gpio/gpio503/value # Key released
```
+ USB HID KeyBoard:
```shell
# Some examples:
# F11:
echo -ne \\x00\\x00\\x44\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # F11 Key pressed
sleep 0.2                                                                                       # Delay
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # F11 Key released

# Up Arrow:
echo -ne \\x00\\x00\\x48\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Up Arrow pressed
sleep 0.2                                                                                       # Delay
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Up Arrow released

# Down Arrow
echo -ne \\x00\\x00\\x50\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down Arrow pressed
sleep 0.2                                                                                       # Delay
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down Arrow released

# Enter KEY
echo -ne \\x00\\x00\\x28\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Enter KEY pressed
sleep 0.2                                                                                       # Delay
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Enter KEY released
```
And you can see all the "HID Usage ID" values [here](https://learn.microsoft.com/en-us/windows/win32/inputdev/about-keyboard-input).I haven't been able to locate the actual HID key values in the USB IF documents. If anyone could point me in the right direction, I would really appreciate it. : )

gregology, 2025-01-10:
Brilliant, thank you. For context, here is my working script to power on the server and navigate a GRUB menu to boot a specific partition.

```bash
# Press power button
echo 1 > /sys/class/gpio/gpio503/value   # Power button pressed
sleep 1
echo 0 > /sys/class/gpio/gpio503/value   # Power button released

# Wait for GRUB menu
sleep 25

# Down arrow
echo -ne \\x00\\x00\\x51\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow pressed
sleep 0.1
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow released
sleep 0.2

# Down arrow
echo -ne \\x00\\x00\\x51\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow pressed
sleep 0.1
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow released
sleep 0.2

# Down arrow
echo -ne \\x00\\x00\\x51\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow pressed
sleep 0.1
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow released
sleep 0.2

# Down arrow
echo -ne \\x00\\x00\\x51\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow pressed
sleep 0.1
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Down arrow released
sleep 0.2

# Enter key
echo -ne \\x00\\x00\\x28\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Enter key pressed
sleep 0.1
echo -ne \\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00 > /dev/hidg0   # Enter key released
```

Now I am going to look at ways to trigger this via an API or alike.

Z2Z-GuGu, 2025-06-13:
> [@Z2Z-GuGu](https://github.com/Z2Z-GuGu) could you explain how to check for the power on LED please?
> 
> thank you for the info so far - some official documentation or a nicer API would be great

Of course, I think the documentation here can help resolve your issue: [https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/development.html](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/development.html). 

I will close this issue for now. If you have any other requests, please feel free to mention them here and @Z2Z-GuGu.

closed 2025-06-13 by Z2Z-GuGu