sipeed/NanoKVM #204: [Feature Request] Swap Memory

ChangHoon-Sung, 2024-12-16:
## Proposal

Enable swap memory creation by default to mitigate low memory situations.

## Background

Currently, NanoKVM easily reaches 100% memory usage when Tailscale is enabled and disk operations are performed. In such cases, the Out-Of-Memory (OOM) Killer terminates processes, potentially leading to server shutdowns or loss of network access. (See #89) To address this issue, Linux swap memory can be utilized. Swap memory leverages disk space as extended, albeit slower, memory.

I acknowledge that using an SD card as the backend storage for swap is slow. However, “functioning slowly” is always better than forced termination of tasks.

Additionally, reducing memory usage in the media area has clear limitations and cannot accommodate future scenarios involving increased memory allocation due to feature additions. Swap memory can act as an effective safeguard in such cases.

## Use Case
I have deployed NanoKVM in a remote overseas location where manual firmware updates are not feasible. My NanoKVM device still operates with 128MB of user memory space and suffers from high memory pressure when using Tailscale to bypass firewalls. To mitigate these issues until updates are possible, I have manually enabled swap memory, which has proven to be highly effective.

## Workaround
I have configured approximately 768MB of swap memory, though swap usage has never exceeded 32MB. A swapfile size of around 64–128MB should suffice. Since the root partition lacks sufficient space, I created the swapfile in the `/data` partition used for storage. Alternatively, future firmware updates could resize the root partition to include the swapfile.

Below is the workaround to configure swap memory:
```
fallocate -l 128M /data/swapfile
mkswap /data/swapfile
swapon /data/swapfile # (Needs to be executed at every boot or added to /etc/fstab)
```
I propose implementing a similar setup as the default behavior for NanoKVM.

ucki007, 2024-12-16:
I see your points / benefits. But I don't think it makes sense to implement that as default. Maybe an option in the UI to enable the swapfile.

Reason: as you already mentioned nanoKVM operates on a SD-Card which are - to my experience - very vulnerable to a lot of writes (which a swapfile would cause).
If it would be implemented as default, I think we would see a lot of SD-Cards dying in no time.

ChangHoon-Sung, 2024-12-17:
@ucki007 That’s a good point! I was focusing solely on the performance of the SD card and didn’t take durability into account. For the reasons you mentioned, enabling this by default would certainly be problematic. As you suggested, exposing the swap option in the Web UI should be sufficient.

ChangHoon-Sung, 2024-12-17:
I will create a new issue requesting the addition of a UI option.

closed 2024-12-17 by ChangHoon-Sung