sipeed/NanoKVM #208: No Internet connectivity on HDMI/HID plugged-in

lemonderon, 2024-12-17:
Whenever I plug-in HDMI or USB HID I lose internet connectivity. The ETH indicator is here, yet no IP Address is being retrieved. 
Basically, if only AUX power and ethernet is plugged in - DHCP is being retrieved / static config works just fine, otherwise - unable to connect to the NanoKVM, but the ETH indicator is being shown and router shows an active link (blinking lights).

I am on the latest 2.1.2 right now. The product was shipped with 2.1.1. Not sure why it differs from the provided here numbering system (1.3.0, 1.2.1).

I have dissembled the item from the case and I see no soldering issues.

lemonderon, 2024-12-17:
I guess it's related to #26 (grounding issue).
Basically, I've tried powering up the device (AUX port) using router USB port and then plugged-in USB HID, HDMI and RJ45. Now it seems to be stable. When I only plug in AUX port to any power source and Ethernet - it always works. When I plug in USB HID to router USB port + ethernet and later add HDMI to the PC - everything works as well and seems to be stable.

lemonderon, 2024-12-19:
Does HID work for you with that cable? When I mod the USB cable like that - HID & power does not work.

lemonderon, 2024-12-20:
So that solution did not work for me (with covering USB connector contacts). 

My setup is rb5009 router that is powered and NAS server. They do not share the same power outlet. If I plug in the Ethernet cable to rb5009 and my NanoKVM - I am unable to use the device properly, unless I use the USB port on rb5009 as HID/Aux power. In this case if i plug in HID port to PC - the device starts losing the ethernet connection which prevents me from using the NanoKVM video feed or just generally reliably connect to it.

Right now, I tried using GL-iNet AR300M16 router (has only 100 mbps ports) as a L2 switch (in bridge mode) between rb5009 and nanoKVM. I am also powering the AR300M16 from USB port on the NAS server. At the same time, the USB HID/power cable i plugged into the same PC. This way everything seems to work reliably for now.

I will write back a little bit later if I have totally resolved my connectivity issues.

mcury1985, 2024-12-20:
It works for me, I have mouse and keyboard control with that modded cable.
I'm powering the device through the AUX, PC-USB modded cable is connected to my desktop and everything just works.

Pin 1 is 5v only, no data goes through it.

lemonderon, 2025-01-09:
So, the issues are gone, and I've bought a shorter 30 cm HDMI. It seems to be working for 2 weeks without any issues. Still a piece of "device" this is.

Waiting for another manufacturer to ship their KVM devices. Hopefully, this time it would be able to work out of the box and without a ton of issues.

tishcufksips, 2025-01-31:
For me, without a USB cable connected to the PC, the device was getting a DHCP IP, but no IP was assigned when the USB was connected to the PC.
With HDMI connected, it was receiving an IP address, but the connection was unstable, and there was no video output on the web console.

I was close to returning it, but after finding this thread and others mentioning grounding issues, I decided to use an FTP cable with metal shielding on the connector (I verified continuity on both ends). So far, it has been working perfectly—no disconnections, and the HDMI output works as well.

I'm not saying this will solve everyone's issues with the device, but it helped with mine.

dotX12, 2025-02-01:
see fix: https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317