sipeed/NanoKVM #210: Add static IP address configuration

klosz007, 2024-12-17:
I understand this KVM is probably dedicated to use with Tailscale and dynamic IPs are non-issue.
But think what happens if DHCP server runs on a computer box controlled by this KVM device.
If this computer is powered off then it might happen that KVM will not be able to obtain IP or renew it.
It must be possible to set KVM to use static IP and not depend on external service (e.g. DHCP).

th3cube, 2024-12-18:
Just read the Manual 

https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html

L00maca, 2024-12-19:
Having it on the GUI would still be nice, even tho I'd argue not an high priority.

klosz007, 2024-12-19:
That's what I meant - I silently assumed that such basic setting (static IP address) should be in GUI. And if it is not there then it's not possible. Indeed I was wrong/my fault - it is possible to set static IP but via CLI only.

jkotran, 2025-01-04:
I set a static IP per the manual. How do I set my DNS server?

th3cube, 2025-01-04:
> That's what I meant - I silently assumed that such basic setting (static IP address) should be in GUI. And if it is not there then it's not possible. Indeed I was wrong/my fault - it is possible to set static IP but via CLI only. The request is then to make it possible via GUI.

I completely agree. There should be an option to set this via the GUI; it would make the process much more user-friendly.

---

> I set a static IP as per the manual. How do I configure my DNS server?

This behavior is unexpected. I thought this would be handled via the `resolv.conf` file located in the `/boot/` directory, which is copied over to `/etc/`. However, that’s not the case. Instead, the file gets overwritten on every reboot, and also hardcoded external DNS servers from locations like China, Google, and Singapore are applied.

Additionally, there seem to be typos in the scripts responsible for handling this, and there is other scripting involved that overrides the expected behavior. A developer should investigate this further. For example, here’s a snippet of the script:

```bash
arping -Dqc2 -Ieth0 $addr || continue
ip a add $inet brd + dev eth0
ip r add default via $gw dev eth0
cat > /etc/resolve.conf << EOF
nameserver $gw
nameserver 8.8.8.8
nameserver 114.114.114.114
EOF
break
```

There should be no hardcoded external DNS servers unless there is a valid reason for it. Instead, the DNS servers provided by DHCP or configured in the static configuration file should be applied. Why hardcode external DNS?

Examples of the hardcoded DNS servers:

```bash
nameserver 8.8.8.8        # Google
nameserver 114.114.114.114 # Singapore?
nameserver 119.29.29.29   # China?
nameserver 223.5.5.5      # China?
```

For now, I’ve set the DNS servers manually in /etc/resolv.conf and made the file immutable using:
`chattr +i /etc/resolv.conf`

This prevents it from being overwritten on reboot. However, this is just a temporary workaround. It would be interesting to hear what the developers have to say about this behavior.

jkotran, 2025-01-04:
@th3cube Thank you for the immutable tip. That's useful.

klosz007, 2025-01-06:
I opened another issue for this DNS issus: https://github.com/sipeed/NanoKVM/issues/245
I suggest you post your findings about DNS there.

xitation, 2025-03-29:
The DNS issue is a typo in the script it creates the wrong file resolve.conf instead of resolv.conf.

PR to fix here - https://github.com/sipeed/NanoKVM/pull/421

adamscybot, 2026-03-02:
It's tangential but I'd advise anyone to ensure their gateway is configured such that any DNS requests are force routed to go to your gateway and and onwards to a DNS server of choice. Maybe a DoH server which applies block lists (controld, mullvad doh, etc). 

Doesnt help for devices that uses DoH directly, but thats not the case here and generally speaking this gets close to the goal of taking control of this acreoss the board.

Its not that uncommon for IoT devices to have preconfigured DNS/NTP servers that include china etc (Phillips Hue Bridge does it too! And that's a hugely popular device). It's cos it needs to work within the bounds of any country-level restrictions for those customers. Not a huge issue but always best as network admin to control all DNS anyways.

dormancygrace, 2026-09-14:
Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

The Web UI has persistent Ethernet DHCP/static IPv4 configuration, including address, subnet mask and gateway validation.

Use the GUI to write the configuration. Legacy hand-edited eth.nodhcp files containing comments still need a parser fix, which I am tracking separately.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/service/network/ethernet.go).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!