sipeed/NanoKVM #211: Feature Request: Enable Crontab

wilsonnkwan, 2024-12-17:
Hi, I understand that Cron is part of the kit, but as of now, the crontabs do not exist, it keeps pointing to var/spool directory but the cron/crontabs do not exist. I tried creating it but it does not survive a reboot. anyway that you can make this permanent or teach me how to do it?

wilsonnkwan, 2024-12-17:
I sort of figured it out, not the most elegant solution, but it works for now.
1. Create a cron file in a more permanent location like /root/ where there is a text file say abc.txt which contains the cron file.
2. Create a initialization file for cron in /etc/init.d/S99Cron, within that file: a) mkdir -p /var/spool/cron/crontabs/ b) cp /root/abc.txt /var/spool/cron/crontabs/root 3) crond (to start the cron).

not sure if it will interfer with the system, so if you have a better solution, do share.

BiLKiNiS, 2026-05-12:
I want to add a crond task to wake on lan the server where my nano KVM PCIe is in to wake up at specific hour to do backup task and then let it go to sleep after
I have a script with the ether wake command that work
I have a crontab -e configured
But I think the crond service is not running and I am not sure how to setup that in a way that would persist after reboot

dormancygrace, 2026-09-19:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2), my independent community firmware for SG2002-based NanoKVM devices, includes a change relevant to this request.

Crontabs have persistent storage under /etc/kvm/cron/crontabs. On the Alpine-based v2 image, /etc/crontabs points there and the base package enables crond through OpenRC. Recreating the volatile spool no longer discards the saved jobs.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/firmware/alpine/packages/nanokvm-base/nanokvm-base.post-install) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/kvmapp/system/init.d/S49persistent-cron).

You're welcome to try **NanoKVM OS v2.0-a2** and let me know how it works on your setup!

[Release and installation details](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2)

Please include your board model, OS version and reproduction steps when reporting results. Thanks!