sipeed/NanoKVM #215: Unable to Access Web Interface - "Test Completed" Message Displayed

jgrubiox, 2024-12-19:
Hello,

I recently purchased a NanoKVM device. When I connect it to my network, the device displays the message "Test Completed!" along with the IP address 192.168.18.89. I am able to connect to the device via SSH, but I cannot access the web interface using the provided IP address.

Here is a photo of the device's screen:
[image]

I can connect to the device via SSH without any issues.
The network is configured correctly, and the IP address is reachable (ping test confirmed).

It seems that the device is stuck in "Test Mode," as indicated by the message on the screen. However, after reviewing the documentation ([user guide](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html)), I could not find any instructions on how to exit "Test Mode" or proceed to the normal operational state.

Could you please assist with resolving this issue? Let me know if you need more details or logs.

Thank you!

Z2Z-GuGu, 2024-12-23:
Hello, I apologize for any mistakes that may have occurred during our shipping process, resulting in outdated devices being sent out. Please use SSH to connect to the OLED at the IP ssh root@192.168.18.89, and execute cat /device_key. Send the result to my email, and I will activate it for you as soon as possible. Once activated, I will notify you, and you will need to log in via SSH and execute this Python script.

[update-nanokvm.py.zip](https://github.com/user-attachments/files/18224758/update-nanokvm.py.zip)

jgrubiox, 2024-12-25:
> Hello, I apologize for any mistakes that may have occurred during our shipping process, resulting in outdated devices being sent out. Please use SSH to connect to the OLED at the IP ssh [...]

Sent email to `z2zmoujietongxue@126.com`

Z2Z-GuGu, 2024-12-30:
> Sent email to `z2zmoujietongxue@126.com`

Hi, I have added the device code. Please try using the updated script above. Thank you!

jgrubiox, 2024-12-31:
> > Sent email to `z2zmoujietongxue@126.com`
> 
> Hi, I have added the device code. Please try using the updated script above. Thank you!

# python update.py
stop service...
killall: kvm_stream: no process killed
rm: can't remove '/tmp/kvm_stream': No such file or directory
OK
start update......
create temporary directory /root/.kvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?[ANONYMIZED]
download firmware done
download lib...
update failed
HTTPSConnectionPool(host='[ANONYMIZED_SERVER]', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?[ANONYMIZED] (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x3fde8975d0>, 'Connection to [ANONYMIZED_SERVER] timed out. (connect timeout=None)'))
killall: kvm_system: no process killed
killall: kvm_stream: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/kvm_stream': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
ping: usage error: Destination address required
OK
load config success
# 192.168.18.89

Not working :(

Ampledd, 2025-01-03:
I have the exact same issue!

jgrubiox, 2025-01-22:
> > Sent email to `z2zmoujietongxue@126.com`
> 
> Hi, I have added the device code. Please try using the updated script above. Thank you!

Still not working, anything new? @Z2Z-GuGu

jgrubiox, 2025-03-03:
Hello, solved with the last firmware!

closed 2025-03-03 by jgrubiox