sipeed/NanoKVM #217: Unstable network connection when switch is set to 100 Mbps

earth4s, 2024-12-20:
I haven't been able to figure this out but I have to force 10 Mbps FDX on my switch to get a stable connection to my NanoKVMs. Ping show random drops and I an unable to access the Web interface when set to auto negotiate or 100 Mpbs.

1) Switch is a Ubiquiti "USW Pro 48 PoE"
2) Used multiple Ethernet Cables, both shielded and unshielded.
3) POE is disabled on the ports used and does not affect stability in any meaningful way
4) Port default is "auto" 10 Mbps/100 Mbps/1 Gbps negotiation
5) When the ports are set to 100 Mbps FDX or HDX the connection is just as unstable as "auto"
6) NanoKVM is flashed with v1.3.0
7) So far affects all the NanoKVMs I have used (I have open only 3 of 5 NanoKVMs that I own)
8) 10 Mbps FDX is the only stable port configuration I have found
9) I have another switch in route (another Ubiquiti switch) that I might be able to test with soon.

Any ideas on what might be the issue on why 100 Mbps is unstable? Or any additional configuration I can test?

archont94, 2024-12-26:
@earth4s could you please check https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2562907472 and verify if this is also the case for you?

avg-I, 2025-01-07:
I am facing a similar issue.
I experience very large response latency unless I force the port to 10M *half*-duplex.
Note that ssh and ping are rather fast with default settings, but video latency is unbearable unless I force the port to lower speed.
Possibly a hardware problem.

shaneaj, 2025-01-18:
I can report a similar issue. I can only connect to the device if i set the switch port to only connect at 10MBPS. However the NanoKVM will actually connect at 10MBS half duplex as well. 

I have swapped network cables and ports and the issue is always present. Another nanokVM i bought does not have this issue.

If I use `ethtool -s eth0 speed 100 duplex full autoneg off` to get the nanoKVM to connect at any other speed it will be very laggy with many errors.

CRCinAU, 2025-01-24:
I have random network failures on 2 of 3 NanoKVM full devices. I've had so many problems with these devices, I'm tempted to just RMA the lot... I managed to get 5 in total, but I'm just not confident at all in their stability...

dotX12, 2025-02-01:
@shaneaj @CRCinAU @earth4s 

see https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2629114317