sipeed/NanoKVM #219: Remote Firmware Update

michaelsage, 2024-12-20:
Hi,
I've got a nanokvm that's in use on a remote site, I am now running into tailscale memory issues having updated to 2.1.2 of the application version. I know this should be fixed in firmware 1.3.0, is there any way to remotely flash or should I send out a new microSD with the 1.3.0 firmware on it?
Thanks

pojlFDlxCOvZ4Kg8y1l4, 2024-12-23:
send out a new SD card

patschi, 2024-12-23:
Why not just using the online feature update? 

Or maybe trying this:
```
cd /etc/kvm/
python update-nanokvm.py
```

michaelsage, 2024-12-23:
> Why not just using the online feature update?
> 
> Or maybe trying this:
> 
> ```
> cd /etc/kvm/
> python update-nanokvm.py
> ```

Because that's for the application not the firmware :( The application is at 2.1.2 and the firmware is at 1.2.0

michaelsage, 2024-12-23:
> send out a new SD card

yeah that's what I thought, I've turned swap on for now

CRCinAU, 2025-01-10:
Is there any difference between updating via the web UI and reflashing the SD card?

I've been hunting for an answer to this for quite a while now...

CRCinAU, 2025-01-10:
Actually - I should document some of my experiments here.... As this Issue seems fitting...

I managed to remotely update the firmware via SSH using the following.

From my desktop:
```bash
$ sudo kpartx -a 20241120_NanoKVM_Rev1_3_0.img
$ mkdir -p /mnt/boot /mnt/rootfs
$ sudo mount /dev/mapper/loop0p1 /mnt/boot
$ sudo mount /dev/mapper/loop0p2 /mnt/rootfs
```

On the NanoKVM, log in via SSH and then do the following - making sure not to reboot from now on, or you will have to reflash the SD manually...
```bash
# rm -f /boot/*
```

Then from the desktop PC again:
```bash
$ scp -a /mnt/boot/* root@<kvm ip>:/boot/
$ sudo rsync -axv --delete-before /mnt/rootfs/ root@<kvm ip>:/
```

On the NanoKVM, ensure that your boot partition looks like:
```
# ls -l /boot/
total 11728
-rwxr-xr-x    1 root     root      11553428 Jan 10 17:31 boot.sd
-rwxr-xr-x    1 root     root        439808 Jan 10 17:31 fip.bin
-rwxr-xr-x    1 root     root             0 Jan 10 17:31 gt9xx
-rwxr-xr-x    1 root     root          3621 Jan 10 17:31 logo.jpeg
-rwxr-xr-x    1 root     root           132 Jan 10 17:39 resolv.conf
-rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.dev
-rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.disk0
-rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.rndis0
-rwxr-xr-x    1 root     root            28 Jan 10 17:31 ver
-rwxr-xr-x    1 root     root             1 Jan 10 17:31 wifi.pass
-rwxr-xr-x    1 root     root             1 Jan 10 17:31 wifi.ssid
-rwxr-xr-x    1 root     root             0 Jan 10 17:31 wifi.sta
```

Make sure `/` looks like:
```
# ls -l /
total 100
lrwxrwxrwx    1 root     root             7 Sep 20 06:00 bin -> usr/bin
drwxr-xr-x    2 root     root         16384 Jan  1  1970 boot
drwxr-xr-x    2 root     root         32768 Jan  1  1980 data
drwxr-xr-x   11 root     root          4780 Jan 10 17:39 dev
drwxr-xr-x   19 root     root          4096 Jan 10 17:39 etc
drwxr-xr-x    6 root     root          4096 Jan 10 17:39 kvmapp
lrwxrwxrwx    1 root     root             7 Sep 20 06:00 lib -> usr/lib
lrwxrwxrwx    1 root     root             3 Sep 20 06:00 lib64 -> lib
lrwxrwxrwx    1 root     root            11 Sep 20 06:00 linuxrc -> bin/busybox
drwx------    2 root     root         16384 Nov 13 01:58 lost+found
drwxr-xr-x    2 root     root          4096 Sep 20 06:00 media
drwxr-xr-x    5 root     root          4096 Sep 20 06:00 mnt
drwxr-xr-x    2 root     root          4096 Sep 20 06:00 opt
dr-xr-xr-x  140 root     root             0 Jan  1  1970 proc
drwx------    3 root     root          4096 Jan 10 17:39 root
drwxr-xr-x    8 root     root           440 Jan 10 17:37 run
lrwxrwxrwx    1 root     root             8 Sep 20 06:00 sbin -> usr/sbin
dr-xr-xr-x   11 root     root             0 Jan  1  1970 sys
drwxrwxrwt    5 root     root           180 Jan 10 17:39 tmp
drwxr-xr-x    7 root     root          4096 Sep 20 06:00 usr
drwxr-xr-x    5 root     root          4096 Sep 20 06:00 var
```

If all goes well, reboot the IP KVM and it should boot, then start loading the app back onto the device. After a little wait, the device should work as normal again but with an updated firmware...

michaelsage, 2025-01-10:
I've got a spare NanoKVM so I'll give this a go. Thank you.

michaelsage, 2025-01-10:
> Is there any difference between updating via the web UI and reflashing the SD card?
> 
> I've been hunting for an answer to this for quite a while now...

Yes one is firmware and one is application although they don't distinguish them very well on the site or documentation

CRCinAU, 2025-01-14:
> @CRCinAU that offline update process is really handy, it really should be documented in the repo somewhere. I was banging my head against a wall trying to get mine to update until I discovered it was due to connectivity to China being blocked on the local network I had the NanoKVM on. I also wasn't waiting long enough for it to complete initial setup, ended up leaving it for 30mins after I moved it to a network where connectivity to China wasn't blocked and now everything is working fine and it auto updated to the latest application version (firmware). Here is a cross-post to another thread if interested with everything I tried though I wish I came across your post above first as an offline update would have been much less hassle! [#186 (comment)](https://github.com/sipeed/NanoKVM/issues/186#issuecomment-2589838447)

Good to see it worked. Feel free to spread the link to the guide - because it was a saviour to me too - as I have mine taped to the side of a rack - and ripping them off to get to the SD card is a royal pain in the ass....

So I improvised hahahah

eljese, 2025-01-16:
> Actually - I should document some of my experiments here.... As this Issue seems fitting...
> 
> I managed to remotely update the firmware via SSH using the following.
> 
> From my desktop:
> 
> ```shell
> $ sudo kpartx -a 20241120_NanoKVM_Rev1_3_0.img
> $ mkdir -p /mnt/boot /mnt/rootfs
> $ sudo mount /dev/mapper/loop0p1 /mnt/boot
> $ sudo mount /dev/mapper/loop0p2 /mnt/rootfs
> ```
> 
> On the NanoKVM, log in via SSH and then do the following - making sure not to reboot from now on, or you will have to reflash the SD manually...
> 
> ```shell
> # rm -fR /boot/*
> ```
> 
> Then from the desktop PC again:
> 
> ```shell
> $ scp /mnt/boot/* root@<kvm ip>:/boot/
> $ cd /mnt/rootfs/
> $ sudo rsync -axv --delete-before . root@<kvm ip>:/
> ```
> 
> On the NanoKVM, ensure that your boot partition looks like:
> 
> ```
> # ls -l /boot/
> total 11728
> -rwxr-xr-x    1 root     root      11553428 Jan 10 17:31 boot.sd
> -rwxr-xr-x    1 root     root        439808 Jan 10 17:31 fip.bin
> -rwxr-xr-x    1 root     root             0 Jan 10 17:31 gt9xx
> -rwxr-xr-x    1 root     root          3621 Jan 10 17:31 logo.jpeg
> -rwxr-xr-x    1 root     root           132 Jan 10 17:39 resolv.conf
> -rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.dev
> -rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.disk0
> -rwxr-xr-x    1 root     root             0 Jan 10 17:31 usb.rndis0
> -rwxr-xr-x    1 root     root            28 Jan 10 17:31 ver
> -rwxr-xr-x    1 root     root             1 Jan 10 17:31 wifi.pass
> -rwxr-xr-x    1 root     root             1 Jan 10 17:31 wifi.ssid
> -rwxr-xr-x    1 root     root             0 Jan 10 17:31 wifi.sta
> ```
> 
> Make sure `/` looks like:
> 
> ```
> # ls -l /
> total 100
> lrwxrwxrwx    1 root     root             7 Sep 20 06:00 bin -> usr/bin
> drwxr-xr-x    2 root     root         16384 Jan  1  1970 boot
> drwxr-xr-x    2 root     root         32768 Jan  1  1980 data
> drwxr-xr-x   11 root     root          4780 Jan 10 17:39 dev
> drwxr-xr-x   19 root     root          4096 Jan 10 17:39 etc
> drwxr-xr-x    6 root     root          4096 Jan 10 17:39 kvmapp
> lrwxrwxrwx    1 root     root             7 Sep 20 06:00 lib -> usr/lib
> lrwxrwxrwx    1 root     root             3 Sep 20 06:00 lib64 -> lib
> lrwxrwxrwx    1 root     root            11 Sep 20 06:00 linuxrc -> bin/busybox
> drwx------    2 root     root         16384 Nov 13 01:58 lost+found
> drwxr-xr-x    2 root     root          4096 Sep 20 06:00 media
> drwxr-xr-x    5 root     root          4096 Sep 20 06:00 mnt
> drwxr-xr-x    2 root     root          4096 Sep 20 06:00 opt
> dr-xr-xr-x  140 root     root             0 Jan  1  1970 proc
> drwx------    3 root     root          4096 Jan 10 17:39 root
> drwxr-xr-x    8 root     root           440 Jan 10 17:37 run
> lrwxrwxrwx    1 root     root             8 Sep 20 06:00 sbin -> usr/sbin
> dr-xr-xr-x   11 root     root             0 Jan  1  1970 sys
> drwxrwxrwt    5 root     root           180 Jan 10 17:39 tmp
> drwxr-xr-x    7 root     root          4096 Sep 20 06:00 usr
> drwxr-xr-x    5 root     root          4096 Sep 20 06:00 var
> ```
> 
> If all goes well, reboot the IP KVM and it should boot, then start loading the app back onto the device. After a little wait, the device should work as normal again but with an updated firmware...
> 
> EDIT: Oh, and to clean up on your PC....
> 
> ```
> $ umount /mnt/boot /mnt/rootfs
> $ kpartx -d 20241120_NanoKVM_Rev1_3_0.img
> $ rmdir /mnt/boot /mnt/rootfs
> ```

Thanks for the guide. Saved so much effort as I was able to update firmware from remote. Just needed to adjust mounts to correct loops.

michaelsage, 2025-01-21:
> Actually - I should document some of my experiments here.... As this Issue seems fitting...
> 
> I managed to remotely update the firmware via SSH using the following.

This is amazing, just did my first remote upgrade. Couple of things to remember...

- If you use a static IP you will need to recreate the eth.nodhcp file (or copy it back) or it will default back to DHCP
- This process (obviously) resets to default passwords
- If you are using tailscale, you will need console access to the nanokvm (other than over tailscale) as you will need to relogin.

I know these are pretty obvious points, but I didn't really think when I did my KVM over tailscale upgrade. Luckily my router has a remote console built in so I managed to recover.

Thank you so much for taking the time to share with the community.

joshuagamboa, 2025-02-04:
any idea if this can be done a Mac machine? kpartx does not exist in the Mac terminal.

michaelsage, 2025-02-04:
> any idea if this can be done a Mac machine? kpartx does not exist in the Mac terminal.

I'd probably just spin up a virtual machine or use a raspberry pi, I had a quick look and I can't see a similar tool!

yodaphone, 2025-02-23:
> Actually - I should document some of my experiments here.... As this Issue seems fitting...
> 
> I managed to remotely update the firmware via SSH using the following.
> 
> From my desktop:

THanks for the guide. was able to do it from WSL on Windows. Worked like a charm

superphreak4, 2025-02-25:
If anyone else has issues with ssh connection being refused. 
I was thinking it was iptables not opening port 22, but it was sshd wouldn't start due to a permission issue on /var/empty

On my first update, I was using my regular user & mounting img under my /home
I redid the remote upgrade process to be safe, doing all desktop commands as root & mounting under /mnt

Z2Z-GuGu, 2025-03-11:
Due to the security concerns raised by many users, SSH is disabled by default in versions above 1.4. You can re-enable it in the web settings or by deleting the file /etc/kvm/ssh_stop on the TF card.

CRCinAU, 2025-03-21:
> To anyone else looking on how to do this: you don't need to mount the image on your local machine and then scp or rsync individual files. Instead, you can upload the entire image to the root of your KVM, mount the partitions, then copy the files locally.

Yep - that'll work too. I did it entirely remotely because I wasn't convinced that the image won't get bigger than the available space on the device as time goes by.

kimnzl, 2025-03-30:
Expanded from @agreenbhm with checksum and backup.
Comments are in brackets () except 1b.

1.  (checksum the source image)
sha256sum 20250217_NanoKVM_Rev1_4_0.img
1b. windows powershell: 
(Get-FileHash 20250217_NanoKVM_Rev1_4_0.img -Algorithm SHA256).Hash.ToLower()
2. (copy the img to the kvm /data folder)
scp 20250217_NanoKVM_Rev1_4_0.img root@<kvm_ip>:/data/
3. (ssh to the kvm)
ssh root@<kvm_ip>
4. (compare to above from step 1, takes 3+ minutes) 
sha256sum /data/20250217_NanoKVM_Rev1_4_0.img
5. (backup existing system)
(backup /boot)
tar c -C / -v -f /data/boot.tar /boot
(backup /)
echo "boot/*" > /data/root_exclude.txt
echo "data/*" >> /data/root_exclude.txt
echo "dev/*" >> /data/root_exclude.txt
echo "mnt/*" >> /data/root_exclude.txt
echo "proc/*" >> /data/root_exclude.txt
echo "run/*" >> /data/root_exclude.txt
echo "sys/*" >> /data/root_exclude.txt
echo "tmp/*" >> /data/root_exclude.txt
echo "var/lock/*" >> /data/root_exclude.txt
echo "var/run/*" >> /data/root_exclude.txt
echo "var/tmp/*" >> /data/root_exclude.txt
echo "var/empty/*" >> /data/root_exclude.txt
echo "var/log/*" >> /data/root_exclude.txt
echo "var/cache/*" >> /data/root_exclude.txt
echo "var/spool/*" >> /data/root_exclude.txt
tar c -C / -v -f /data/root.tar -X /data/root_exclude.txt /
6. (mount the image to loopback)
losetup -f -P /data/20250217_NanoKVM_Rev1_4_0.img
7. (confirm where is mounted) 
losetup -a
8.  (assuming it mounted the image to /dev/loop0p1 and /dev/loop0p2): 
mkdir -p /tmp/boot && mkdir -p /tmp/root
9. (mount the partitions from the image)
mount /dev/loop0p1 /tmp/boot && mount /dev/loop0p2 /tmp/root
10. (copy updated /boot)
cd /tmp/boot && cp -av * /boot/
11. (copy updated /)
cd /tmp/root && cp -av * /
12. (unmount the image partitions)
cd / && umount /tmp/boot && umount /tmp/root
13. ('eject' the image)
losetup -d /dev/loop0
14. (remove the update img)
rm /data/20250217_NanoKVM_Rev1_4_0.img
15. (reboot the kvm)
reboot
16. (in the webui) update the app

rbm78bln, 2025-04-21:
Okay, maybe I'm a bit old-fashioned... but I justs did this on my NanoKVM:
> sync && echo /dev/mmcblk0p* | xargs -n1 mount -o remount,ro && echo u > /proc/sysrq-trigger
> iptables -F && nc -l -p 8989 | dd bs=1M iflag=fullblock oflag=direct of=/dev/mmcblk0

And on my host machine I did this:
> unxz -fc < 20250217_NanoKVM_Rev1_4_0.img.xz | dd bs=1M iflag=fullblock status=progress | nc -v -v -n 192.168.1.8 8989

So I guess with a little bit of fiddling around one could also easily do this without any netcat involved:
> curl 'https://github.com/path/to/someimage.img.xz ' | unxz -fc | dd bs=1M iflag=fullblock oflag=direct of=/dev/mmcblk0

... of course, don't forget to (violently) ro-mount your stuff first.

Then simply reboot and you're fine to go:
> echo b > /proc/sysrq-trigger

patschi, 2025-05-08:
Very much inspired by the work of those comments:
https://github.com/sipeed/NanoKVM/issues/219#issuecomment-2583355791
https://github.com/sipeed/NanoKVM/issues/219#issuecomment-2741278285

# Download

```shell
wget https://github.com/sipeed/NanoKVM/releases/download/v1.4.0/20250217_NanoKVM_Rev1_4_0.img.xz
xz --decompress --keep 20250217_NanoKVM_Rev1_4_0.img.xz
```

# Upload/Preparation

```shell
export NK_IP="192.168.0.36"
export NK_RPW="root"
export NK_IMG="20250217_NanoKVM_Rev1_4_0.img"

SSHPASS=$NK_RPW sshpass -e scp -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null $NK_IMG root@$NK_IP:/nanokvm.img
SSHPASS=$NK_RPW sshpass -e ssh -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null root@$NK_IP
```

# Overwrite (on NanoKVM)

```shell
(losetup -a | grep -q loop7) && echo "[!] Make sure loop7 is free and not mounted! Do not proceed."

losetup -P -r /dev/loop7 /nanokvm.img
mkdir -p /tmp/boot /tmp/rootfs
mount /dev/loop7p1 /tmp/boot
mount /dev/loop7p2 /tmp/rootfs

rsync -axv --delete-before /tmp/boot/ /boot/
rsync -axv --delete-before /tmp/rootfs/ /rootfs/

umount /tmp/boot
umount /tmp/rootfs
losetup -d /dev/loop7
rm /nanokvm.img

reboot
```

## Error when updating /boot?

```shell
# rsync -axv --delete-before /tmp/boot/ /boot/
building file list ... done
boot.sd
rsync: [receiver] write failed on "/boot/boot.sd": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(380) [receiver=3.2.7]
rsync: [sender] write error: Broken pipe (32)
# rm /boot/*
```

Re: @rbm78bln 

> Okay, maybe I'm a bit old-fashioned... but I justs did this on my NanoKVM:

I like your approach. But wouldn't that also overwrite `/data` with all ISOs?

> PS: But yes, of course, a ready-to-use script by the developers and a button on the UI would be fancy. 😎

I think it would be best having two separated partitions with the image, and one partition (as an overlay) with NanoKVM-application and/or config...