sipeed/NanoKVM #223: Very long load of web UI

archont94, 2024-12-25:
I just started to use NanoKVM.
When web UI will load, I'm pretty happy about perfomance and features. 
The problem is that it is not always fully loading - sometimes I need to refresh page multiple times before I will even get chance to give my password. I checked with `htop` - its not CPU load or RAM consumption, its about 5% of CPU and 35MB or RAM, so it should be fine. 

If I will leave browser open for long time, page will eventually load and there will be password pop-up, but its unreliable. In this case it took over 9 minutes to load page.

[image]

How can I fix it?

fenio, 2024-12-26:
Which browser are you using? For me Firefox is completely unreliable with NanoKVM. Chrome works just fine.

archont94, 2024-12-26:
I'm using Chrome.

archont94, 2024-12-26:
Ok, so I did some additional steps to identify reason for long download of js files.

Direct rsync copy on NanoKVM is as fast as micro SD card, as expected:
[image]

rsync copy over SSH is very slow - this indicate that the problem is not only related to `NanoKVM-Server`, but even direct copy is causing problems:
[image]

scp without encription should be faster, but it isn't:
[image]

I was trying to check how long it will take to transfer file with nc (tcp and udp), but those never finished (they always stopped after around 293k transfered, which was weird).

I also noticed, that ping response from i.e. 1.1.1.1 is taking a long time, about 20-50ms.
`mtr` shows about 18%-27% of packet loss to host in local network. I don't have problems like that with any other device in my network (I checked `mtr` on different hosts in same network, 0% packet loss).
During all of my tests, CPU usage was about 5-8%, a lot of RAM was available. I even stopped KVM-Server service to make sure it is not causing problems. `iostat` show that disk is idle in 96%, so it should not be a problem.
[image]

This makes me think that the problem is with network stack configuration, not KVM-service itself - if its not possible to reliably transfer data using rsync, scp or nc, then it should not be surprising that KVM-service also cannot serve files quickly.
Fast rsync transfer on local disk of NanoKVM suggest that the problem is not with CPU performance (i.e. with encrypting data), but rather with communication between hosts.

Is there anything what can be done on image to increase network stack reliability?

archont94, 2024-12-26:
There are multiple dropped RX packets:
[image]

archont94, 2024-12-27:
I logged traffic with `tcpdump -i eth0 -w /data/capture.pcap` and I see a lot of TCP retransmissions on Ethernet, but none on RNDIS.

As extra debug step, I added small ethernet switch between NanoKVM and my main switch to see if this will act differently, and for some reason it does - I don't see TCP retransmissions anymore.
I checked with `ethtool eth0` what are differences, but the only one I see is `Link partner advertised pause frame use: Symmetric` on not-working setup vs `Link partner advertised pause frame use: Symmetric Receive-only` on working one.

It would suggest that the problem is with physical layer of ethernet. @Z2Z-GuGu is this known error? I don't mind having 100Mb device, but if this device isn't able to negociate proper ethernet settings or if there are signal integrity issues in product, then its is a big no-no. Do you know any workaround for this problem? I cannot have extra ethernet switch for all of my NanoKVMs.

archont94, 2024-12-27:
Ok so after additional testing seems like the rootcause is the same as in https://github.com/sipeed/NanoKVM/issues/26

If I power NanoKVM from powerbank and I have only ethernet attached, the speed of ethernet is as expected. 
In the moment I connect HID USB or HDMI, ethernet connection issues are appearing again. Connecting UART GND to power common ground doesn't help, in my case the only solution is to use additional ethernet device (either switch or PoE injector) which implements proper galwanic isolation.

@Z2Z-GuGu you mentioned that this is [capacitive isolation](https://github.com/sipeed/NanoKVM/issues/26#issuecomment-2333366615), but it wasn't implemented properly in NanoKVM device - whats the point of having ethernet magnetics, if you connect all ground together?

My current understanding is that this issue cannot be solved via software, as hardware itself isn't implemented properly. Will sipeed provide additional ethernet isolator for customers who experienced this problem, or will provide reworked hardware version, which implements isolation properly?

archont94, 2024-12-27:
After another few hours of debugging, changing ethernet cables to different types (cat 5, cat 5e, cat6, cat6a), with power bank or direct power supply I cannot establish ethernet link anymore, not even with additional isolation devices.

I think that this is proof that NanoKVM ethernet physical layer implementation is bad - right now I'm not sure if the problem is in grounding, wrong ethernet cable type (unlikely as I tried multiple ones), or maybe devide itself is broken.

Z2Z-GuGu, 2024-12-30:
I sincerely apologize for the negative experience you’ve encountered. This issue indeed stems from a design oversight on our part. The likelihood of this problem occurring is relatively low, which has made it challenging for us to test a fully effective isolator.

In October 2024, our company briefly faced a connectivity issue with Ethernet. We urgently tested a solution using an external PoE isolator combined with a PoE isolator connected to UART GND, which seemed to resolve the issue. However, the Ethernet connectivity problem then unexpectedly disappeared, and we have not encountered it again since.

We are currently working on testing effective measures in various challenging power environments, and I will keep you updated on our latest progress here. Thank you for your understanding.

archont94, 2024-12-30:
> However, the Ethernet connectivity problem then unexpectedly disappeared, and we have not encountered it again since.

If you have possibility, please try with PicoPSU - I use MeanWell HDR-100-12 power supply and basic no name 90W PicoPSU. MOBO is N100 based 6 sata 4 Ethernet device, which can be found on AliExpress. 

What are next steps for me? Can I perform some extra verification of my NanoKvm which will help you? Do you have other version of software (i.e. with changes in link detection)? Do you want me to send it back for detailed investigation to see why it's not able to setup Ethernet link anymore? Will I get refund or working device in exchange?

archont94, 2025-01-07:
@Z2Z-GuGu aby updates?

maximko, 2025-01-12:
Does nanokvm lite has the same issue? That's a serious deal braker.

archont94, 2025-01-12:
> Does nanokvm lite has the same issue?

For 99% it does, as it's the same PCB design.

medsandor, 2025-01-20:
I have received today the nano kvm lite, also have ~15% packet loss, entering some keys and it appears on the screen after 10-15, sometimes 20 secs later. This is really unusable. The server where I connected is a standard home PC, ATX PSU after an APC UPS. Can the UPS interfere here (but the UPS is a must for my server)? Ground wiring is properly existing in the house, the UPS also passes it through to the PC. Is it possible to request that mentioned isolator?

medsandor, 2025-01-21:
Update: I have a POE switch for my IP cameras, I have removed one camera, and put the nano KVM lite there. And everything is working, fast, no connection drops, no packet loss anymore from ping. But I need that port on the POE switch for my camera :) So really need some advise, isolator, anything you are talking about to make this KVM work on any switches. The other switch is a 16 port TP-LINK manageable switch where the KVM performs poorly.

medsandor, 2025-01-29:
Also just noticed today when tried to use Nano KVM lite, that it was very slow again. Ping again shows packet losses, but it was still on the previously working POE switch. Unplugged from USB, plugged back, and now it is fast again. I have noticed it is very hot, cannot hold it in hand too long. Room temperature is around 25 Celsius degree. Not sure when it will slow down again.

dotX12, 2025-02-01:
@archont94, @medsandor, @wieluk 
I couldn't figure this problem out for 24 hours.....
Only thanks to a random review on aliexpress I found a post that a person had a similar problem with the device losing packets / js loading is very long or no screen at all / video lag is 30 seconds or the device can't get an address from DHCP at all. 

All the person in the review did was to replace the internet cable with a shielded cable (with metal contacts). I followed his advice and bought a shielded cable, plugged it in and what was my surprise that everything worked. The delay disappeared, h.264 and mjpeg started working. 
Also you should forcibly set your router to 100 mbit and turn off speed auto-negotiation.

avg-I, 2025-02-01:
@dotX12 it might have been my comment.
I also shared that information in one of the open issues here.

dotX12, 2025-02-01:
@avg-I, its u?)))

[image]

kustusch, 2026-01-08:
This KVM is really picky about cables indeed.
I connected mine through 30ft of CAT6 cable and it was dropping packets every 20-30 seconds and terminal was freezing for several seconds. Lowering speed to 10Mbps made connection stable but FPS went down to around 15.
I could not replace cable with shielded since it is in the wall. 
I ended up connecting small 8 port switch at location where KVM is installed and then connected KVM through short 6ft CAT6 cable, works fine at 100Mbps but I actually tried several cable and found only one out 5 working! All cables were tested and work perfectly fine with other devices. 
Very strange. I love this little thing but network interface needs work.

AT-StephenDetomasi, 2026-03-12:
Unreal... I just tried one of my NanoKVMs with a cable that has metal shielding on the connector, and it actually solved my intermittent connectivity. Whyyyy is this even a thing??

Evg33, 2026-08-25:
In my case, a 1m shielded cables didn't solve the problem.
I solved the problem using a short unshielded cable (30cm).

gamburger2005, 2026-09-24:
I'm recevired NanoKVM PCI with wifi and have same issue. But when I use wifi connection all works good/

hungjuchen666-cpu, 2026-10-29:
Same issue here on a **NanoKVM Cube** — adding data in case it helps.

**Hardware / firmware**
- NanoKVM Cube, application 2.5.2, kernel 5.10.4-tag- (riscv64, Buildroot)
- Target host connected via HDMI + USB (HID + RNDIS `usb0`)
- Ethernet: 100Mb/s Full, autoneg on

**Symptom**
`eth0` loses carrier for 1-2 s at irregular intervals of a few hours, then comes back:

```
Oct 28 01:23:01 kernel: bm-dwmac 4070000.ethernet eth0: Link is Down
Oct 28 01:23:02 kernel: bm-dwmac 4070000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Oct 28 10:17:40 ... Link is Down
Oct 28 13:00:16 ... Link is Down
Oct 29 11:24:30 ... Link is Down
Oct 29 15:23:54 ... Link is Down
```

- `/sys/class/net/eth0/carrier_changes`: 8 after ~20 h uptime; 16 after ~24 h on the next boot.
- Every drop kills long-lived TCP sessions through eth0 (SSH port forwards break, SMB mounts over them go dead).
- RX dropped counters: `eth0` 51087 vs `usb0` 98.
- The USB RNDIS link (`usb0`) to the same target host carried an SSH session that stayed up for 20+ hours without interruption over the same period.

**What was ruled out**
- Cable and switch port: replaced with a new cable, then moved to the exact cable and switch port that a NanoKVM Pro had been using without any drops. Drops continue on the Cube.
- Client side: the peer's NIC shows no link change at the drop times, and the peer machine never sleeps.

Is there a recommended fix for the Cube (isolator, specific cable type, firmware setting), or should this unit be considered for RMA?