sipeed/NanoKVM #224: How to use "Custom Scripts"? I wish to send CTLR-ALT-DEL combo

TheLinuxGuy, 2024-12-26:
Please add documentation to the wiki about "custom scripts" functionality, it is unclear:
- what type of file to upload?
- how to create a script that sends keyboard combination CTLR-ALT-DEL (i'm shocked this isn't in UI like pikvm and other common keyboard shortcuts... this really should be UI added but I digress - this question is about using script to do the same)

fenio, 2024-12-26:
You can fine some details in another issue: https://github.com/sipeed/NanoKVM/issues/199

roondar, 2025-01-15:
It would be good to add it directly in the menu bar ? @Z2Z-GuGu

wj-xiao, 2025-04-08:
Added in version [2.1.6](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#216-6eb4a4e-2025-02-14).

closed 2025-04-08 by wj-xiao