sipeed/NanoKVM #227: NanoKVM presence adds a default network interface on fresh ubuntu installs which make the machine unreachable

EcceGratum, 2024-12-26:
I installed a fresh Kubuntu 24.04 on a machine which has NanoKVM plugged all the time.
I installed from a usbkey, not from NanoKVM virtual image feature.

I noticed on the first launch of the new install, that there was 2 network interfaces instead of one. One is for the physical LAN on the motherboard as expected and the other one appears to be interface to NanoKVM.
The network interface to NanoKVM is the one used by default which makes the machines unreachable from LAN and it can't connect to internet.

In my case, the network interface name is enxda248c27027c . 
I am trying to remove that extra interface but in the meantime i want to let people know about the issue

archont94, 2024-12-27:
This is expected and can be disabled:
[image]

EcceGratum, 2024-12-27:
@archont94 Thank you

closed 2024-12-27 by EcceGratum