sipeed/NanoKVM #228: Unable to Change Keyboard Layout and Missing "@" Symbol

the-ab, 2024-12-27:
Hello,

I am encountering an issue with the keyboard layout on my system. When using the NanoKVM device, I am unable to change the keyboard layout, and as a result, the "@" symbol is not working properly.

Problem:
I cannot change the keyboard layout to the desired language (e.g., German or US layout), and the expected symbols are not typed correctly.
The "@" symbol is missing, and instead of typing "@", the " symbol is typed.
This issue persists on both physical and on-screen keyboards, with the on-screen keyboard outputting the wrong character (i.e., " instead of @).

Expected Behavior:
I would like to change the keyboard layout to a standard layout (e.g., German or US) and expect the "@" symbol to work correctly when pressing the corresponding key.

Could you please assist me in resolving this issue, or let me know if there is a specific configuration required to enable the correct keyboard layout?

Thank you!

marcpopp, 2025-01-09:
Having similar Problems.

NanoKVM is connected to Windows 10 PC with German Keyboard Layout. Working on a MacBook.

Problems with keyboard layout found so far:
* Special characters not working:  \ or |
* Pasting from MacBook to Windows uses US Keystrokes on Windows.

stuartm, 2025-01-12:
+1

Being unable to type | for a UK(GB) Layout makes doing pretty much anything infinitely harder in Linux shell.

cybe, 2025-01-12:
There has been some related changes two weeks ago that might address these issues: https://github.com/sipeed/NanoKVM/commit/26078fe46e43d4543d7b09901b4992e4fbe4f01f.

the-ab, 2025-01-12:
it only works if I set the keyboard layout to US on the client, client keyboard layout to German not working, it would be nice if the layout on the nanoKVM could be changed

marcpopp, 2025-01-16:
Updated to Version 2.1.5 and now the Backslash \ is working. Still no Pipe | ...

MatejKovacic, 2025-01-19:
The same with me. Also, Slovenian characters (č, š, ž,etc.) are working. But no pipe `|`, which is critical.

archef2000, 2025-01-26:
I have the same problem with the special character "^" in the german layout. Shifted number keys work

MHOOO, 2025-01-29:
Same here. I have the nanokvm attached to a device that uses neo layout (like dvorak, but with a focus on german). Unfortunately I am unable to enter any keys besides the ones on the default layer...

mcury1985, 2025-01-30:
Same problem here, but for pt-br.
"/" not working.

v7o73, 2025-02-06:
I use a Swiss German Layout on my Laptop.
If I use the virtual Keyboard or the copy/paste window from nanoKVM, it translates it as a US Keyboard an AltGr characters dosen't work.
So I can't send complex Passwords to login.

nickbee22, 2025-02-12:
I'm also having this issue with backslash \ becoming hash #
Version 2.1.5 - EN-GB keyboard layout

zentarim, 2025-02-12:
Got he same issue.
"|" is replaced by "#" what makes it extremely difficult to work with Linux environment.

Software version 2.1.5

ucki007, 2025-02-19:
Still an issue with new firmware (1.4.0) and software (2.2.0)

MikeinGH, 2025-02-21:
I can confirm: the problem persist with newest firm- and software. Please please fix it

brkai, 2025-03-10:
I discovered that the problem seems to be related to the browser:
when i use firefox or ungoogled chromium (chromium itself should also be OK) everything ist fine,
i can use the so called german umlaute and have my german keyboard layout on the host computer (laptop, wich has also a german keyboard). the only thing, that doesn't work remotely, is the number-block
i've tried LibreWolf and falcon (versions: flatpak):
voilla, the prob is there!

.... didn't dig deeper, due to working hours ;-)

Deses, 2025-03-10:
I have an Spanish layout keyboard and if I type with it directly, symbols and special characters are sent correctly, but anything that needs Alt+Gr doesn't get sent, which could be that the browser or OS is catching that input and not sending it to the NanoKVM webui. OK, no problem, that's what the Virtual keyboard is for, right?

The issue is that the virtual keyboard is in the en-US layout and doesn't match my language nor my keyboard, and we've never got asked for the layout to begin with, so of course... but! Pressing the corresponding keys in the virtual keyboard sends the correct real keyboard key, but they don't match visually.
For instance: I press the button right to the 0 in the VKB, which is a dash - and underscore _ and the key sent is a question mark ?, which is the correct one in the real KB.

TL;DR: While using a real keyboard the keys send the correct characters, but the virtual keyboard doesn't match.
@wj-xiao the WebUI should autodetect the kb layout from the browser/OS and update the virtual keyboard AND we should given the option to manually change the layout.

zentarim, 2025-03-10:
I wonder if that this guy was fighting with same keyboard-related issues.

https://danielhb.github.io/article/2019/05/06/noVNC-QEMU-RFB.html

lvanoverberghe, 2026-02-17:
This basic issue is still there.