sipeed/NanoKVM #229: WebUI Password Reset Steps Undocumented

aerodb, 2024-12-27:
Using a Nano KVM I have locked myself out of the WebUI. 

Issue: 
There is no documentation that I can find that details the steps to reset the WebUI password. Default(webui) is admin/admin but SSH uses root/root as login. 

Further the steps provided on the WebUI indicate to navigate to a pwd file, which is not preset on my unit when connected via SSH.

ucki007, 2024-12-29:
Hi @aerodb ,

that is exactly, how it works (or at least should work).

If the file "/etc/kvm/pwd" does not exist, the default user/password combination (admin/admin) for the WebUI should apply.

So it is documented - but you say, it does not work for you? That should be a bug then.

As one of my new NanoKVMs did not work with the default credentials, I had to follow the instructions when it was delivered a few weeks ago. Worked as described - but I don't remember the version my NanoKVM was delivered with.
And I tried it to check if the current version is somehow affected (Image 1.3.0, WebGUI 2.1.3) - also works as described.

Could you please provide the output of "ls -la /etc/kvm/" when logged in via SSH?

ucki007, 2025-01-22:
Hi @aerodb ,

if you also have yourself locked out of SSH (root user account), there is only one option:

Reflash the sd card with the latest image from here: https://github.com/sipeed/NanoKVM/releases

aerodb, 2025-01-22:
@ucki007 Would there happen to be a guide somewhere or is this is simple as copying to the SD card? Thank you for the guidance and support thus far.

aerodb, 2025-01-22:
To any finding my issue and seeking the guide, I found this. https://en.wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html

@ucki007  please correct me if this is the wrong resource.

ucki007, 2025-01-23:
Hi @aerodb ,

all fine - that is the official source.