sipeed/NanoKVM #230: Device overheats, becomes unavailable (PCIe version)

jtagcat, 2024-12-28:
1. Keep console open for 20 minutes
2. Device becomes unavailable: from Tailscale and LAN.

LAN SSH stays available. Device does not recover without removing power. When trying to use again right away, it dies after two minutes.

jtagcat, 2024-12-30:
<details>
<summary>Possibly fault in frequent polling hogging the CPU?</summary>

```
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.524772]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.529783] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.529794] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.636768] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.636768]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.641660] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.641671] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.748770] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.748770]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.754000] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.754011] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.860785] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.860785]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.866365] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.866396] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.972845] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.972845]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.978635] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177972.978646] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177973.084790] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177973.084790]  
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177973.090981] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:57 kvm-9111 kern.warn kernel: [177973.090993] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.196820] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.196820]  
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.793472] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.793483] <<-GTP-INFO->> Guitar reset
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.900794] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:56:58 kvm-9111 kern.warn kernel: [177973.900794]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.427367] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.427379] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.532864] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.532864]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.533348] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.533357] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.640878] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.640878]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.645858] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.645869] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.752879] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.752879]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.758198] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.758209] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.864882] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.864882]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.870539] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.870550] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.976888] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.976888]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.982884] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177977.982895] <<-GTP-INFO->> Guitar reset
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177978.088886] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177978.088886]  
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177978.095207] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:57:02 kvm-9111 kern.warn kernel: [177978.095218] <<-GTP-INFO->> Guitar reset
[... cut ...]
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178118.969551] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178118.969562] <<-GTP-INFO->> Guitar reset
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178119.075622] <<-GTP-ERROR->> I2C transfer error. errno:-121
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178119.075622]  
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178119.081901] <<-GTP-ERROR->> I2C Read: 0x814E, 10 bytes failed, errcode: -121! Process reset.
Dec 30 04:59:23 kvm-9111 kern.warn kernel: [178119.081913] <<-GTP-INFO->> Guitar reset
```

</details>

jtagcat, 2024-12-30:
Found Tailscale hogging the memory, while it itself was dead. Tailscale is known to be a big blob on low-end devices. Possibly could be worked around with using swap.

jtagcat, 2025-01-02:
> Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this issue.

HID issues are likely by bad design, the usb-c on the card needs to be in a specific position (hot glue it).

Krusenbaum, 2025-01-08:
I am having the same issue. Came with latest firmware, worked for 20-30 min and now refuses any connection via port 80 UI.

Krusenbaum, 2025-01-08:
This is without any changes. Received the unit today, installed it, logged into the UI, checked for updates (none where found) and had one KVM session that I left open. Device is now just pingable and I can log into it using SSH (but it does not seem to accept my user / password combination). As I cannot reach the device now, I have no idea what is enabled / disabled.

Krusenbaum, 2025-01-08:
Update: I SSHed into the device using root:root, disabled the tailscale service:

mv /etc/init.d/S98tailscaled /etc/init.d/S98tailscaled.disabled

Rebooted the device and now the webinterface is working again.

Krusenbaum, 2025-01-08:
I manually rebooted the device via SSH and it seems to be working for now...weird

wj-xiao, 2025-01-16:
It should be Tailscale was killed due to OOM.
Please check #149 .

ssttevee, 2025-03-23:
I have still been experiencing this issue using image v1.4.0 and application version 2.2.4. It happens on only two of my many (10+) units, one pcie and one cube, so I think it might have to do with the networking in this particular location. AFAIK, the only difference in this location is that the nanokvm is on a tagged vlan behind a managed switch.

Additionally, I have monitored the memory on `htop` and `free -m` multiple times while the web ui crashes without going over 50% RAM usage. Either it's not related to OOM, or i have a completely different issue.

As a workaround, I've noticed that if I manually run `/tmp/server/NanoKVM-Server` from ssh, it does not crash again. (FYI, run `setsid /tmp/server/NanoKVM-Server` instead if you want to close your terminal)