sipeed/NanoKVM #232: RMA request - Contact options?

ucki007, 2024-12-28:
Hi all,

did anyone of you have direct contact with sipeed support in the last few week?

I have a defective ATX module delivered (dead on arrival) with one of my NanoKVM cubes and tried to get in contact with sipeed via mentioned contact options (wl@sipeed.com, support@sipeed.com). Got no reply at all - not even a "received your request notification".

So ... does anybody have an idea on how to get in touch with sipeed directly (other than the two email contacts mentioned above)?

wj-xiao, 2025-01-16:
Sorry for that. Has anyone replied to you yet?

AJolly, 2025-01-20:
No ones replied to me, and its been a week.  Have been unable to get any device to send hdmi video signals with my new one :/

ucki007, 2025-01-20:
Hi @wj-xiao ,  @AJolly ,

forgot to update here.

My first try of contact (from my gmx mail) never got a reply - tried again with another email provider (googlemail). That worked and I got a reply within a very short time.

So maybe there is an issue with some email providers.

For me everything is good - Replacement-Kit is on the way.

closed 2025-01-20 by ucki007