sipeed/NanoKVM #233: Lights out mode

bilogic, 2024-12-30:
[image]

Possible to turn off all LEDs?

umbertoragone, 2025-03-23:
I have the NanoKVM Lite, and I managed to disable the blinking blue activity light (the bottom right one in the picture) by running the following command in the NanoKVM terminal (either via SSH or by accessing the terminal from the web UI):

`echo 0 > /sys/class/leds/led-user/brightness`

The command forces that LED brightness to 0 (turns it off), but upon reboot of the NanoKVM it turns back on (as in blinking when there is activity).

The "5V" blue LED (top left) of the non-Lite version and red power LED (bottom left) seem to be directly tied to the 5V0 rail (as per the schematic), so they can't be turned off in software, and would require hardware modifications to permanently disable them.

I hope they add a toggle in the settings to at least override the default blinking behaviour of the blue activity LED.

Neurofibromin, 2025-06-27:
On the pcie version this works (until restart)
```
echo none > /sys/class/leds/mmc1::/trigger
echo none > /sys/class/leds/mmc0::/trigger
echo none > /sys/class/leds/led-user/trigger
```