sipeed/NanoKVM #237: Question

myxxmikeyxx, 2025-01-01:
Hello, what Linux OS is it running? I was wondering if I ssh in could I run a custom package? I wanted to maybe run something like Cloudflared locally on it so I can pass it through to my Cloudflare 0 trust tunnel and still be able to get to it externally. Unless you have something better you recommend or if you don't recommend that at all that is fine, I have openvpn setup but half the time is doesnt work right.

ucki007, 2025-01-01:
Hi @myxxmikeyxx ,

as you can see from /etc/os-release, it is a custom image built using Buildroot:

`NAME=Buildroot
VERSION=-g9c961a8be-dirty
ID=buildroot
VERSION_ID=2023.11.2
PRETTY_NAME="Buildroot 2023.11.2"`

I think, I read that somewhere in the documentation aswell but could not find it right now.

Skywalker8510, 2025-01-16:
Hey @myxxmikeyxx Did you ever figure out how to get cloudflared on there. I was thinking out getting one but it wouldn’t do me any good without being able to forward with cloudflare.

myxxmikeyxx, 2025-01-18:
@ucki007 Thank you. I didn't see that but I am not surprised to hear that.

@Skywalker8510 no not yet. if it is a custom build of Buildroot then that's basically an Arduino. (I know its not really but you can't just install things and change things without doing a custom Buildroot build and add in what you want.)
I have done a custom Buildroot in the past for a raspberry pi that only needed to do one job but that was a pain. Lots of struggle. I see they have a Buildroot docker container but have not tried that. I might one long weekend try and make my own but I would need to see if I can find their Buildroot config first.