sipeed/NanoKVM #239: USB does not work on certain computers

Sergiio, 2025-01-02:
I recently purchased NanoKVM and used it on a Mac mini, it worked correctly.

The problem was when connecting it to a Dell Precision 3660 desktop and a Lenovo desktop.

On both computers, I get an image but the mouse-keyboard does not work, as if it were not connected. The Dell runs Windows and Linux, the same thing happens on both systems. In the case of the Lenovo, it has Ubuntu.

I have updated NanoKVM to the latest version, the same thing happens.
I have changed the USB for several high quality ones that I had at home, the same thing happens.

Exactly the same thing happens to my friend, we have the same Dell model.

ucki007, 2025-01-02:
Maybe related to #183 or #190 ?

I know - both topics handle UEFI / BIOS but maybe there is some relation?

Sergiio, 2025-01-02:
> Maybe related to #183 or #190 ?
> 
> I know - both topics handle UEFI / BIOS but maybe there is some relation?

I have tried this, unfortunately without success.

nuvolacat, 2025-01-02:
Hi, I'm your friend.

I tried creating the BIOS file and now it doesn't give me the KEYBOARD NOT FOUND error anymore but I still can't use the keyboard.

It's a new Dell Precision 3460 with Xpenology. The keyboard doesn't work on the Xpenology boot screen...

I don't know if there are any other options to try.

Thank you very much.

ucki007, 2025-01-02:
> Hi, I'm your friend.
>
> I tried creating the BIOS file and now it doesn't give me the KEYBOARD NOT FOUND error anymore but I still can't use the keyboard.
>
> It's a new Dell Precision 3460 with [...]

Hi @nuvolacat,

can you try to enter UEFI/BIOS using a real keyboard and check, if your mouse works there? Than we have the same behaviour as I have in #190. Maybe that helps to get hands on that issue.

nuvolacat, 2025-01-02:
Wow! The mouse in UEFI/BIOS setup works fine.

Well, either it's a specific problem with DELL or maybe with some NanoKVM firmware update it can be solved.

Hopefully they're aware of the problem.

Z2Z-GuGu, 2025-01-04:
Hey guys, I think I've found the root of the problem! Some specific computer models need the USB DP/DN pins to be at a low level for a while before they can recognize new USB devices. This lets them enter USB detection mode. But with self-powered devices like NanoKVM, this low-level time is often overlooked, which is why some hosts (like DELL) can't detect when a USB device is plugged in.
It’s not just in the BIOS; some systems disconnect USB devices after going to sleep. Since NanoKVM skips that low-level time and tries to reconnect right away, those systems also miss the chance to connect the devices.
We tested on one of the problematic machines at the company and found that adding a low-level delay before starting USB is effective. We're planning to include this feature in the 2.1.4 release scheduled for next Thursday. I think this should help resolve the issues you’ve been facing.
Before that, you can use the following script to check if this approach works.
```shell
echo > /sys/kernel/config/usb_gadget/g0/UDC
sleep 1
ls /sys/class/udc/ | cat > /sys/kernel/config/usb_gadget/g0/UDC
```
Looking forward to hearing back from everyone!

MarcelSM24, 2025-01-04:
I'm facing the same problem with an HP elitedesk g3 800 mini. The script you just provided it's like disconnecting and connecting the nanoKVM right?  Because already tried this with the script and doing it physically, and it's not working for me. 
Only fails in BIOS/UEFI in OS is working great. 
For more testing when I should execute the script?

MarcelSM24, 2025-01-04:
> Hey guys, I think I've found the root of the problem! Some specific computer models need the USB DP/DN pins to be at a low level for a while before they can recognize new USB devices, and lets them enter USB detection mode. But with self-powered devices like NanoKVM, this low-level time is often overlooked, which is why some hosts (like DELL) can't detect when a USB device is plugged in. It’s not just in the BIOS; some systems disconnect USB devices after going to sleep. Since NanoKVM skips that low-level time and tries to reconnect right away, those systems also miss the chance to connect the devices. We tested on one of the problematic machines at the company and found that adding a low-level delay before starting USB is effective. We're planning to include this feature in the 2.1.4 release scheduled for next Thursday. I think this should help resolve the issues you’ve been facing. Before that, you can use the following script to check if this approach works.
> 
> ```shell
> echo > /sys/kernel/config/usb_gadget/g0/UDC
> sleep 1
> ls /sys/class/udc/ | cat > /sys/kernel/config/usb_gadget/g0/UDC
> ```
> 
> Looking forward to hearing back from everyone!

Hi @Z2Z-GuGu, 
I'm facing the same problem with an HP elitedesk g3 800 mini. The script you just provided it's like disconnecting and connecting the nanoKVM right?  Because already tried this with the script and doing it physically, and it's not working for me. 
Only fails in BIOS/UEFI in OS is working great. 
For more testing when I should execute the script?

nuvolacat, 2025-01-04:
> Hey guys, I think I've found the root of the problem! Some specific computer models need the USB DP/DN pins to be at a low level for a while before they can recognize new USB devices, and lets them [...]

In my case it still doesn't work. In BIOS/UEFI setup mode I can use the mouse without problems but the keyboard doesn't work.

I have tried it on my two Dell servers and the exact same thing happens on both. The only progress is the fix proposed at https://github.com/sipeed/NanoKVM/issues/190

ucki007, 2025-01-05:
Hi @Z2Z-GuGu ,

I tried your script in the following combinations with my Optiplex 3000

- with "/boot/BIOS" in place
- without "/boot/BIOS" in place
- executing your script before booting the Optiplex
- executing your script, when already in BIOS
- executing your script, when booted to installed OS (Ubuntu 24.10 in this case)

Results:
1. None of the combinations enables Keyboard in BIOS/UEFI (still only Mouse is available, when "/boot/BIOS" is in place)
2. Executing your script if Host OS is already booted leads to keyboard and mouse losing functionality. Only mouse returns, when resetting input devices via NanoKVM WebGUI. Restarting the NanoKVM then returns to normal functionality in Host OS.

So there is no improvement for my topic (#190 )

Let me know if I can test anything further.

nuvolacat, 2025-01-10:
> Hi @Z2Z-GuGu ,
> 
> I saw the update to 2.1.4 being available today and did the update.
> 
> Result: still no improvement for #190
> 
> What about you @Sergiio & @nuvolacat - do you see an improvement?

It doesn't work for me. It's a shame, really.

Sergiio, 2025-01-16:
> Hi [@Z2Z-GuGu](https://github.com/Z2Z-GuGu) ,
>
> I saw the update to 2.1.4 being available today and did the update.
>
> Result: still no improvement for [...]

Sorry, I've been away from home for a few days. 

I have updated NanoKVM and it works on Windows. In the BIOS I don't have a keyboard or mouse. But it does seem to work on Windows. Its something!

asychev, 2025-02-27:
I have similar issues. Very often keyboard works in BIOS for some time and then just stop. Reboot of NanoKVM sometimes helps, sometimes does not at all. Reset HID does not work. With /boot/BIOS or without. The issue is mostly with all sorts of N100/N150 boxes.

Z2Z-GuGu, 2025-03-26:
I found during further testing that some motherboards' BIOS do not favor HID devices having an IAD descriptor (a descriptor for composite devices). Therefore, in the new version 2.2.5, I added 'HID Only' mode. To test it, I purchased an inexpensive HP EliteDesk 800 G3, and the new mode runs well on this motherboard's BIOS. I have also received positive feedback from several users regarding the new mode. Everyone can enable 'HID Only' in the mouse icon of version 2.2.5. I look forward to your feedback. Thank you!

ucki007, 2025-03-26:
> I found during further testing that some motherboards' BIOS do not favor HID devices having an IAD descriptor (a descriptor for composite devices). Therefore, in the new version 2.2.5, I added 'HID [...]

Tested it on my Dell Optiplex - works fine! I will close the corresponding issues

Thanks for solving!

Z2Z-GuGu, 2025-04-01:
I have received feedback from many people who have successfully used it, so I think it's time to close this issue. And if anyone has any questions, feel free to create a new issue and @ me.

closed 2025-04-01 by Z2Z-GuGu