sipeed/NanoKVM #240: update failed

Ampledd, 2025-01-03:
Hi,

I got into the web UI once, but when I try to click check for updates, I get kicked out of the http session and it becomes unresponsive.

I tried to re-flash the SD. Didn't work, then I tried the python updater:

# python update-nanokvm.py
stop service...
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
start update......
create temporary directory /root/nanokvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1735905894
download firmware done
download lib...
update failed
HTTPSConnectionPool(host='maixvision.sipeed.com', port=443): Max retries exceeded with url: /api/v1/nanokvm/encryption?uid=d1cc44a91359c262 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x3fdae3b6d0>, 'Connection to maixvision.sipeed.com timed out. (connect timeout=None)'))
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
cp: can't stat '/kvmapp/kvm_system': No such file or directory
cp: can't stat '/kvmapp/server': No such file or directory
/etc/init.d/S95nanokvm: line 76: /tmp/kvm_system/kvm_system: not found
/etc/init.d/S95nanokvm: line 85: /tmp/server/NanoKVM-Server: not found
OK

Looks like the maixvision.sipeed.com is down, which is driving me crazy at this point because I can't get it to work.
Any help is greatly appreciated!

jkotran, 2025-01-04:
I successfully updated yesterday. I experienced a problem similar to the original poster's. I could access cdn.sipeed.com. I could NOT access maixvision.sipeed.com.

 In my case I was behind a web proxy that blocks access to Chinese hosted web sites. I had to temporarily bypass the web block to successfully update.

gregoreau, 2025-01-14:
I had the same issue as @jkotran and was able to fix it the same way, here is a cross-post to another issue with more details if interested as I also ended up leaving it for 30mins after initial boot to it could set it itself up properly, I don't think I was waiting long enough initially:
https://github.com/sipeed/NanoKVM/issues/186#issuecomment-2589838447