sipeed/NanoKVM #241: HDD LED &  POWER LED NOT WORKING

writzthecat, 2025-01-03:
Hey Guys,

just received my NanoKVM in the full Version.

Got everything connected, ATX Daughterboard installed and the NanoKVM is working, but the HDD LED and POWER LED in my case are now dead.

Used the correct pins on the daughterboard. Any idea ?

stvorl, 2025-01-11:
They are LEDs, so might you have the polarity reversed? There is easy to misconnect that chaotic colored pigtails.

As far, as I can see, KVM-B unit is having LEDs' IN and OUT pins just connected together (while switch pins connected through 212G optocouplers, which controlled by NanoKVM). 

So you may try to connect case LED's pigtails directly to motherboard pins, while reset and power remains connected via KVM-B.

roondar, 2025-01-17:
https://github.com/sipeed/sipeed_wiki/blob/main/docs/hardware/en/kvm/NanoKVM/quick_start.md#wiring

ckotte, 2025-03-19:
What benefit do I have with HDD lights connected to the KVM-B board? Can I see HDD activity in the web UI or somewhere? I read somewhere that HDD (status) monitoring is not supported with NanoKVm Cube. Does it only works with the PCIe version?

sinyb, 2025-08-19:
Also, the power LED status should (optionally) refresh faster, 10 Hz or even more.
I have just saw, for the first time, that ASUS PRIME motherboards use power LED for announcing boot status: slow blinking around 0.5 Hz during POST, fast blinking (4 Hz) for memory error, super slow blinking (~0.2 Hz) for boot error, maybe there is more... 
Of course, having HDD LED in UI would be soooo nice, but it also needs to refresh even faster :)
I agree that power LED color was not chosen very well, full bright green circle would be much better IMO, and of course a bright red one for HDD (Yes, my first PC had those colors, with a yellow TURBO :) )