sipeed/NanoKVM #243: Improvement Request for Static IP Configuration File Parsing

NightowlKr, 2025-01-03:
**Issue:**
script file"/etc/init.d/S30eth" fails to handle comments in "/boot/eth.nodhcp"

**Suggested fix:**
Add a filter to ignore comments in the `/boot/eth.nodhcp` file, using:

`cat /boot/eth.nodhcp  |  grep -v '^#'  | while read inet gw`

Proposed code file "/etc/init.d/S30eth" change:
```sh
start() {
    printf "start ethernet: "
    if [ -e /boot/eth.nodhcp ]
    then
        [ -e /boot/eth.nodhcp ] &&
            cat /boot/eth.nodhcp | grep -v '^#'  | while read inet gw
            do
                addr=${inet%/*}
                netid=${inet#*/}
                [ -z $gw ] &&
                gw=$( echo $addr| ( IFS='.' read a b c d; echo $((
                    (((((($a<<8)+$b)<<8)+$c)<<8)+$d)
                        & (((1<<$netid)-1)<<(32-$netid))
                    ))
                )) &&
                gw=$(($gw>>24&0xff)).$(($gw>>16&0xff)).$(($gw>>8&0xff)).$((1+( $gw>>0&0xff )))

                arping -Dqc2 -Ieth0 $addr || continue
                ip a add $inet brd + dev eth0
                ip r add default via $gw dev eth0
                cat > /etc/resolve.conf << EOF
nameserver $gw
nameserver 8.8.8.8
nameserver 114.114.114.114
EOF
                break
            done &&
            ip a show dev eth0|grep inet || (
                udhcpc -i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid &>/dev/null
                ip a show dev eth0|grep inet
            ) || (
                # failed to apply dynamic addr, need a available static addr to visit the LAN
                inet=$RESERVE_INET
                addr=${inet%/*}
                ip a add $inet brd + dev eth0
            ) || exit 1
    else
        (udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid) &
    fi

    echo "OK"
}
```

archont94, 2025-01-04:
I think it would be even better to add `yq` binary and read configs from YAML files.

dormancygrace, 2026-09-19:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2), my independent community firmware for SG2002-based NanoKVM devices, includes a change relevant to this request.

Both the Ethernet startup script and the Web API parser strip comments from eth.nodhcp and handle empty lines and CRLF. Full-line and trailing comments no longer get interpreted as address or gateway fields.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/kvmapp/system/init.d/S30eth) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/server/service/network/ethernet.go).

You're welcome to try **NanoKVM OS v2.0-a2** and let me know how it works on your setup!

[Release and installation details](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2)

Please include your board model, OS version and reproduction steps when reporting results. Thanks!