sipeed/NanoKVM #244: Slow Upload of Images using SCP

scyto, 2025-01-05:
I have a nanokvm pcie+poe+wifi

On Ethernet i get upload speeds of ~1.2MB/s

On 2.4gz WIFI5 i get 2.5MB/s

Of 5ghz WIFI6 (why do you make it so hard to join a pure wifi6 network, i needed to learn to use wpa_cli to figure it out) i also get 2.5MB/s even for a HT40 1x1 MIMO device this is incredibly slow and i am getting radio connection speeds in the 230 Mb/s range.

In all 3 scenarios i would expect to upload at 10MB/s not 2.5MB/s

HTOP showed i had memory and CPU to spare during the transfer

Are my expectations wrong? 
If not what should i check

archont94, 2025-01-06:
@scyto I would check transfer speed of files between different partitions on NanoKVM (to eliminate problems with SD card itself or i.e. high CPU usage) and later would check speed via RNDIS USB Ethernet connection. There are known issues with Ethernet on NanoKVM Cube, but my understanding is that they should not happen on PCIe version.

scyto, 2025-01-07:
@archont94 

I have the USB header on the nanokvm pcie connected  to the motherboard header, copying files from client  > host webui that lets me see the USB storage device was super fast, so defintely rules out SD card

I am not sure what you mean van RNDIS USB ethernet connection - how would i go about testing that?

I guess this means the same issue exists on the nanonkvm pcie for both ethernet and wifi

for reference, my version numbers:

```
Image Version
v1.3.0

Application Version
2.1.3
```

archont94, 2025-01-07:
> I am not sure what you mean van RNDIS USB ethernet connection - how would i go about testing that?

https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/updating.html#Check-via-USB-RNDIS-Network-Interface

scyto, 2025-01-07:
> > I am not sure what you mean van RNDIS USB ethernet connection - how would i go about testing that?
> 
> https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/updating.html#Check-via-USB-RNDIS-Network-Interface

i already read that, it covers a windows device attached to the external USB port

the usb on my pice card at the back is non functional because the header on the card is connected to the linux host machine i am trying to control - there is no way to plug a windows machine into the card via USB and the linux host does seem to see a USB network device of any kind

does this mean i would need to disconnect the internal header and reconnect it to the backplate to test?  given the speed of direct gadget access from linux host to the kvm i am pretty cetain i will see same speeds as its just usb with differentl applications protocol

seems like something is seriously broken in the nanonkvm ethernet/wifi stack tho....

archont94, 2025-01-07:
If it's connected with HID USB to Linux machine it should be even easier, NanoKVM should create virtual USB network adapter there. Instructions overs it as well. If it's not visible on your Linux host (check with lsusb), enable it in NanoKVM webui.

> seems like something is seriously broken in the nanonkvm ethernet/wifi stack tho....

That's what I'm trying to help you troubleshoot - for me RNDIS was working perfectly fine, but the problem was with regular Ethernet connection (and it was confirmed by Sipeed that it's bad hardware design). But it was on Cube, not PCIe version.

scyto, 2025-01-07:
already enable in the kvm, usb0 appears perfectly

on the linux host it uses a buildroot derived OS and lsusb doesn't return much useful

```
$ lsusb -v
Bus 003 Device 005: ID 1a86:8091
Bus 003 Device 001: ID 1d6b:0002
Bus 001 Device 001: ID 1d6b:0002
Bus 003 Device 004: ID 05e3:0610
Bus 003 Device 006: ID 3346:1009
Bus 003 Device 008: ID 0bda:8179
Bus 003 Device 009: ID 09eb:0131
Bus 004 Device 001: ID 1d6b:0003
Bus 004 Device 002: ID 174c:3074
Bus 003 Device 003: ID 1a86:8091
Bus 002 Device 001: ID 1d6b:0003
Bus 003 Device 007: ID 09ea:0130
Bus 003 Device 002: ID 174c:2074
```
it seems to see the mass storage device on USB 3-7, but not much else
```
$ dmesg | grep usb
[    0.629611] usbcore: registered new interface driver usbfs
[    0.629611] usbcore: registered new interface driver hub
[    0.629614] usbcore: registered new device driver usb
[    0.691593] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    0.691597] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.691598] usb usb1: Product: xHCI Host Controller
[    0.691599] usb usb1: Manufacturer: Linux 6.6.48 xhci-hcd
[    0.691600] usb usb1: SerialNumber: 0000:00:0d.0
[    0.691838] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.06
[    0.691839] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.691840] usb usb2: Product: xHCI Host Controller
[    0.691841] usb usb2: Manufacturer: Linux 6.6.48 xhci-hcd
[    0.691842] usb usb2: SerialNumber: 0000:00:0d.0
[    0.694278] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.06
[    0.694279] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.694280] usb usb3: Product: xHCI Host Controller
[    0.694281] usb usb3: Manufacturer: Linux 6.6.48 xhci-hcd
[    0.694282] usb usb3: SerialNumber: 0000:00:14.0
[    0.695952] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.06
[    0.695954] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.695955] usb usb4: Product: xHCI Host Controller
[    0.695956] usb usb4: Manufacturer: Linux 6.6.48 xhci-hcd
[    0.695957] usb usb4: SerialNumber: 0000:00:14.0
[    0.696441] usbcore: registered new interface driver usblp
[    0.696448] usbcore: registered new interface driver usb-storage
[    0.696458] usbcore: registered new interface driver usbserial_generic
[    0.696461] usbserial: USB Serial support registered for generic
[    0.700748] usbcore: registered new interface driver usbhid
[    0.700749] usbhid: USB HID core driver
[    0.934760] usb 3-3: new high-speed USB device number 2 using xhci_hcd
[    1.064204] usb 3-3: New USB device found, idVendor=174c, idProduct=2074, bcdDevice= 0.01
[    1.064232] usb 3-3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    1.064246] usb 3-3: Product: AS2107
[    1.064254] usb 3-3: Manufacturer: ASMedia
[    1.064262] usb 3-3: SerialNumber: USB2.0 Hub
[    1.176969] usb 4-1: new SuperSpeed USB device number 2 using xhci_hcd
[    1.192590] usb 4-1: New USB device found, idVendor=174c, idProduct=3074, bcdDevice= 0.01
[    1.192618] usb 4-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    1.192629] usb 4-1: Product: AS2107
[    1.192638] usb 4-1: Manufacturer: ASMedia
[    1.192645] usb 4-1: SerialNumber: USB2.0 Hub
[    1.304740] usb 3-7: new high-speed USB device number 3 using xhci_hcd
[    1.432160] usb 3-7: New USB device found, idVendor=1a86, idProduct=8091, bcdDevice= 1.21
[    1.432187] usb 3-7: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    1.432200] usb 3-7: Product: USB HUB
[    1.513743] usb 3-3.1: new high-speed USB device number 4 using xhci_hcd
[    1.613822] usb 3-3.1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=70.60
[... 25 lines cut ...]
[    2.189389] usb 3-3.1.3: Manufacturer: Generic
[    2.189396] usb 3-3.1.3: SerialNumber: 00000002
[    2.297747] usb 3-9: new high-speed USB device number 8 using xhci_hcd
[    2.426482] usb 3-9: New USB device found, idVendor=0bda, idProduct=8179, bcdDevice= 0.00
[    2.426510] usb 3-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.426523] usb 3-9: Product: 802.11n WLAN NIC
[    2.426554] usb 3-9: Manufacturer: Realtek
[    2.489712] usb 3-3.1.3.6: new full-speed USB device number 9 using xhci_hcd
[    2.576140] usb 3-3.1.3.6: New USB device found, idVendor=09eb, idProduct=0131, bcdDevice= 0.02
[    2.576168] usb 3-3.1.3.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.576181] usb 3-3.1.3.6: Product: Virtual HID
[    2.576190] usb 3-3.1.3.6: Manufacturer: Generic
[    2.576198] usb 3-3.1.3.6: SerialNumber: 00000002
[    2.588433] input: Generic Virtual HID as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.1/3-3.1.3/3-3.1.3.6/3-3.1.3.6:1.0/0003:09EB:0131.0004/input/input7
[    2.640860] hid-generic 0003:09EB:0131.0004: input,hidraw3: USB HID v1.10 Keyboard [Generic Virtual HID] on usb-0000:00:14.0-3.1.3.6/input0
[    2.646446] input: Generic Virtual HID Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.1/3-3.1.3/3-3.1.3.6/3-3.1.3.6:1.1/0003:09EB:0131.0005/input/input8
[    2.646798] input: Generic Virtual HID System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.1/3-3.1.3/3-3.1.3.6/3-3.1.3.6:1.1/0003:09EB:0131.0005/input/input9
[... cut ...]
[    2.700042] hid-generic 0003:09EB:0131.0005: input,hidraw4: USB HID v1.10 Mouse [Generic Virtual HID] on usb-0000:00:14.0-3.1.3.6/input1
[   10.701477] usb 3-9: RTL8188EU rev D (TSMC) romver 0, 1T1R, TX queues 2, WiFi=1, BT=0, GPS=0, HI PA=0
[   10.701482] usb 3-9: RTL8188EU MAC: 90:de:80:51:c5:49
[   10.701483] usb 3-9: rtl8xxxu: Loading firmware rtlwifi/rtl8188eufw.bin
[   10.703463] usb 3-9: Firmware revision 28.0 (signature 0x88e1)
[   11.151502] usbcore: registered new interface driver rtl8xxxu 
```

scyto, 2025-01-07:
i will install generic ubuntu and see what happens

auanasgheps, 2025-01-11:
I have this issue as well, but with the NanoKVM Full.

Ethernet is negotiated at 100 Mb/s but the transfer is only about 1.5 MB/s. Is it a limitation of the SoC?

archont94, 2025-01-11:
> limited by the CPU

Not true. Check transfers via RNDIS - 12-18MB/s. The problem is fucked design around Ethernet PHY, causing instability of Ethernet connection. Look for amount of TCP retransmissions.

scyto, 2025-01-17:
> causing instability of Ethernet connection

I am not seeing rety issues and i am seeing the same issue over the pci wifi edition.... i believe there is more than one issue here  (or that the root cause of this issue and the retries you see is something else)

scyto, 2025-02-22:
revisiting, i am not seeing retyrs on the physical Ethernet transfers

i suspect whatever the issue is it is causing the retries you are seeing @archont94 and not that the retrys are the issue (just a symptom)

i agree there is definitely something effed with the design for both physical ethernet and wlan

scyto, 2025-02-22:
definitely see retries when pushing the connection a little with iperf3 (this is all wired testing), this IMO isn't fit for purpose

```
alex@truenas:~$ iperf3 -c 192.168.1.147
Connecting to host 192.168.1.147, port 5201
[  5] local 192.168.1.33 port 51152 connected to 192.168.1.147 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  7.10 MBytes  59.6 Mbits/sec   66   53.7 KBytes       
[  5]   1.00-2.00   sec  6.28 MBytes  52.7 Mbits/sec   12   45.2 KBytes       
[  5]   2.00-3.00   sec  6.34 MBytes  53.2 Mbits/sec   12   42.4 KBytes       
[  5]   3.00-4.00   sec  7.15 MBytes  60.0 Mbits/sec    7   52.3 KBytes       
[  5]   4.00-5.00   sec  7.33 MBytes  61.5 Mbits/sec   12   50.9 KBytes       
[  5]   5.00-6.00   sec  7.64 MBytes  64.1 Mbits/sec   17   45.2 KBytes       
[  5]   6.00-7.00   sec  6.90 MBytes  57.9 Mbits/sec   11   41.0 KBytes       
[  5]   7.00-8.00   sec  6.03 MBytes  50.6 Mbits/sec   12   41.0 KBytes       
[  5]   8.00-9.00   sec  7.15 MBytes  59.9 Mbits/sec   11   50.9 KBytes       
[  5]   9.00-10.00  sec  7.39 MBytes  62.0 Mbits/sec   14   50.9 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  69.3 MBytes  58.1 Mbits/sec  174             sender
[  5]   0.00-10.00  sec  68.8 MBytes  57.7 Mbits/sec                  receiver
```

Zepan, 2025-02-22:
hi, are you testing while video transfering？maybe mjpeg streaming cost most bandwidth?

scyto, 2025-02-24:
> hi, are you testing while video transfering？maybe mjpeg streaming cost most bandwidth?

good suggestion, just tested without, still ~1.5MB/s with SCP

iPerf3 a lot better

```
Connecting to host 192.168.1.147, port 5201
[  5] local 192.168.1.33 port 44708 connected to 192.168.1.147 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  13.0 MBytes   109 Mbits/sec    1    270 KBytes       
[  5]   1.00-2.00   sec  11.2 MBytes  94.4 Mbits/sec    0    313 KBytes       
[  5]   2.00-3.00   sec  10.6 MBytes  89.1 Mbits/sec    0    338 KBytes       
[  5]   3.00-4.00   sec  11.4 MBytes  95.4 Mbits/sec    3    201 KBytes       
[  5]   4.00-5.00   sec  10.6 MBytes  89.1 Mbits/sec    0    239 KBytes       
[  5]   5.00-6.00   sec  11.4 MBytes  95.9 Mbits/sec    0    273 KBytes       
[  5]   6.00-7.00   sec  10.6 MBytes  88.6 Mbits/sec    0    303 KBytes       
[  5]   7.00-8.00   sec  11.3 MBytes  94.9 Mbits/sec    0    331 KBytes       
[  5]   8.00-9.00   sec  11.5 MBytes  96.4 Mbits/sec    1    266 KBytes       
[  5]   9.00-10.00  sec  10.8 MBytes  90.7 Mbits/sec    0    301 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   112 MBytes  94.4 Mbits/sec    5             sender
[  5]   0.00-10.03  sec   111 MBytes  93.0 Mbits/sec                  receiver

```

so why is SCP so slow?

Zepan, 2025-02-24:
Strange, I just tested scp, get 4~5MB/s.
Your iperf test result shows the hardware is OK.
Maybe there have other software comsume the cpu? did you update to newest FW?
[image]

scyto, 2025-02-24:
yeah on latest frimware

i tested SCP from WinSCP, i tested SFTP from the app Termius

remeber i am testing the ethernet port, not the wifi connection.... the wifi was faster - which just shouldn't be the case....

scyto, 2025-02-24:
> Strange, I just tested scp, get 4~5MB/s.

I would also say this is far to low for a 100Mbit Fast Ethernet connection, it should be *double*

If i had to guess its that the device can't cope with the encryption algorothims being negotiated, i have yet tested to see of the sshd on the device has been compiled with none (which is what would be required to test this thesis)

scyto, 2025-02-24:
oh i should add the best i got with SFTP last night, with no nano kvm services running was ~3MB/s - this is still way too low.

Zepan, 2025-02-25:
Hi, I think 4~5MB/s is reasonable, as TF card write speed is about 11MB/s,  so the theoretical speed is 1/(1/11+1/12.5)=5.8MB/s

We just tested in NanoKVM Pro, it is about 10X faster, as it is 1000M ethernet and onbaord eMMC:
[image]

scyto, 2025-02-25:
> Hi, I think 4~5MB/s is reasonable, as TF card write speed is about 11MB/s, so the theoretical speed is 1/(1/11+1/12.5)=5.8MB/s

I don't think the TF card speed has anything to do with it give the speed i am able to copy over USB to the same location - which is   much faster.  also have NEVER hit 4MB/s to 5MBss.  unless when you say TF card you don't mean the SD card (which i swapped out for high speed one as part of my testing)

Also what is the Nano KVM Pro and will it have a PCI edition?

and for a gigabit card that's an awfully slow transfer too, it should be in the 100MB/s range with an eMMC and gigbit connection - easily, i genuinely think you have a fundamental issue somewhere here

give the low CPU specs i suspect either encryption is happening on CPU - remember the speed via USB to the SD card on the noraml KVMs

a dd test of the SD card as exposed by the USB device:

```
root@ZimaCube:/media/sda-usb-Linux_File-Stor_# dd if=/dev/zero of=/media/sda-usb-Linux_File-Stor_# oflag=direct bs=128k count=16k
16384+0 records in
16384+0 records out
2147483648 bytes (2.1 GB, 2.0 GiB) copied, 1.91942 s, 1.1 GB/s
```
happy to perform another test but overall the speed seems fine of the SD card....

Zepan, 2025-02-26:
hi, GbE+emmc around 50-60MB/s is reasonable, emmc is not as fast as you think, it is about 100-150MB/s in CM4, ~250MB/s in CM5, due to soc limit

in our board, it is about 120MB/s,
so the total speed is 1/(1/120+1/120)~60MB/s

[image]

scyto, 2025-02-26:
> due to soc limit

I should be able to get whatvever the uncached sustatin speed is - on the Normal Nano  this appears to be ~50MB so not a limitations of the SD card interface

We also have proven that at the network level a pure iperf3 is capbale of ~54MB/s
I would expect to be able to get more than 5MB/s in this scenario.

This leaves the following possiblities:
- indeed a inherent SOC limitation when trasferring in-memory objects from the network driver to the file system driver (low RAM speeds)
- an inherent SOC limitation caused by the encryption / compression uses by SCP or SFTP (not surprsing in a low powered SOC)

tl;dr this is an SOC limitation and thus is what is (though i still think if you compiled your image with support for encryption=none that would be an interested test, but no need to do that, if you ever opne source your build system for the base OS i am happy to play and figure it out

closed 2025-02-26 by scyto