sipeed/NanoKVM #245: DNS servers (re)set automatically to some wird settings after each device reboot

klosz007, 2025-01-06:
Hi,

I have IP address/gateway set statically.

I noticed that DNS servers in /etc/resolv.conf are set to some weird settings after each reboot, as follows:
nameserver 192.168.0.1
nameserver 8.8.4.4
nameserver 8.8.8.8
nameserver 114.114.114.114
nameserver 119.29.29.29
nameserver 223.5.5.5
(first is some 'local/private' IP but not from my local subnet, two next servers are Google's, the last three are unknown to me),

None of these servers would work in my case because my IoT network segment is only allowed to send DNS requests to my internal DNS servers, in my local/private subnet.

If /etc/resolv.conf is changed manually, it works fine then but it restores itself to incorrect values after every device reboot. Same about changing /boot/resolv.conf.

I found a wrokaround in another issue report (modify /etc/resolv.conf and then set immutable flag on it with chattr) and then it works as expected (changes to /etc/resolv.conf survive reboot). But it is really not as it should work.

stvorl, 2025-01-09:
I found that NanoKVM dhcp client creates /etc/resolv.conf.dhcp with proper DNS server and domain suffix, obtained from DHCP server. But it now used, unless copied or symlinked to /etc/resolv.conf.
So, after some experiments, and using your workaround, I created /etc/init.d/S99resolv-fix:
```
#!/bin/sh
mkdir -p /run/network

case "$1" in
  start)
        echo "Fixing resolv.conf to DHCP version"
        chattr -i /etc/resolv.conf
        rm /etc/resolv.conf
        cp /etc/resolv.conf.dhcp /etc/resolv.conf
        chattr +i /etc/resolv.conf
        ;;
  stop)
        ;;
  restart|reload)
        "$0" stop
        "$0" start
        ;;
  *)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
esac
exit $?
```
Sure, it's quite crude and optimistic solution, but it works.
However, I can't predict what consequences this will lead to. Probably, the default settings were made as is for some purposes, not just for fun.

markuman, 2025-02-06:
Alternativley, override the `/etc/resolv.conf` and make it immutable: https://github.com/markuman/nanokvm-mitigations/blob/latest/cleanup.yml#L24-L31

wj-xiao, 2025-02-18:
Fixed in the version [2.1.6](https://github.com/sipeed/NanoKVM/releases/tag/2.1.6) . Please refer to [CHANGELOG.md](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md) for details.

wj-xiao, 2025-02-18:
Also check the #311 .

gshpychka, 2025-02-18:
> Fixed in the version [2.1.6](https://github.com/sipeed/NanoKVM/releases/tag/2.1.6) . Please refer to [CHANGELOG.md](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md?rgh-link-date=2025-02-18T03%3A22%3A41.000Z) for details.

@wj-xiao so should this be closed?

wj-xiao, 2025-02-18:
@gshpychka Ok. Thanks for the reminder.

closed 2025-02-18 by wj-xiao

xitation, 2025-03-29:
The DNS issue is a typo in the script it creates the wrong file resolve.conf instead of resolv.conf.

PR to fix here - https://github.com/sipeed/NanoKVM/pull/421