sipeed/NanoKVM #246: Keyboard Repeats randomly

djaychela, 2025-01-06:
I'm getting random keyboard repeats, as if the keys are being held down (they are not).  It's happening randomly and making everything difficult.  Running on a system which works fine if I plug a real USB keyboard into it (even viewing via the KVM).  Clearly something is wrong here, but I can't see any settings which may help.  See screenshot.
[image]

Craftefix, 2025-01-09:
Same issue, need a fix for that; even the copy function gets it wrong.
And it's not random (in my case), it's always, and it has a random length.

Craftefix, 2025-01-09:
[image]
[image]

Craftefix, 2025-01-09:
this is what it looks like on a Pi 3b+, its the newest software

talha5389, 2025-03-19:
Similar issue happening to me as well. 

1. HID/Keyboard was not working
2. Restarting HID not worked
3. Rebooting KVM worked however after I had closed off remote session, ctrl key was reported as pressed. Pressing left ctrl/right ctrl keys on actual keyboard didn't reset state of ctrl key. I had to plug they HID usb cable, then pressing left/right ctrl key resetted key states and after that I was able to plug nanokvm back in

This has happened at least 3-5 times before to me

komw, 2025-03-30:
Same here, keys repeats randomly, its hard to write anything directly on the keyboard, delay over local network is terrible... newest firmware 2.2.5.. Also speed of local disk is terrible, writing an small ISO (hundred of MB) over network is taking a very long time (I know that its a 100mbit ethernet, but it goes like over 10mbit)

komw, 2025-03-31:
Ok, it seems that I found a reason for that, but still I don't know why it happens. When I directly connect nanoKVM to my router the problem exists, and also I found that pinging the  nanoKVM device causes about 20% packet loss. I dont know why it happens, as the cable is 100% ok (0,5m of cat5e cable, without any problems with my cable tester), and I'm pinging the KVM directly form my router. I got about 20% of ping loses, and problems with repeating keys, etc.
But If I reconnect the nanoKVM into my switch, using the same cable, everything works fine. There is 0% packet loss, and also there is no any problems with repeating keys etc. 
Also changing port in my router to another one didnt helps. Connected to router => problems with pings/repeaded keys, connected to switch => everything is ok.

JordiGrasvi, 2025-09-02:
I have the same issue with firmware 2.2.9