sipeed/NanoKVM #247: Tailscale logging out

svedsvedsveden, 2025-01-06:
I have had my nanokvm (full), since the middle of december. Before updating tailscale worked flawlessly.  But now after updating bot nanokvm and tailscale firmware - the nanokvm keeps logging out of tailscale on its own.  Sometimes the tailscaleconnection on the nanokvm last for many hours sometimes just minutes. But every time I try to login from outside my LAN the tailscaleconnection crashes. When tailscale crashes, I can still login in to nanokvm on LAN connection - and when I check tailscale again, it prompts me with "The device has not been bound yet. Please login and bind this device to your account."

Does anyone have the same issue, and does anyone know of a fix, other than downgrading tailscale to an older version?
My conclusion is that I am having the same memory issues that other people are having. But I am not techsavy enough to find out on my own.

NanoKVM firmware 2.1.3
Tailscale firmware 1.78.1

diamondiron, 2025-01-07:
I can't get tailscale to be stable at all, period. Do you have steps for downgrading the tailscale version?

svedsvedsveden, 2025-01-07:
Steps for downgrading is in this thread [](https://github.com/sipeed/NanoKVM/issues/149)

Unfortunately I can't figure out how to execute the steps. And then again, downgrading the tailscale firmware, could also be a security risk, so I'm a bit hesitant about doing it anyway.

I am hoping the next firmware update will have a proper fix for this issue

wj-xiao, 2025-01-16:
Please check #149 . We will discuss the relevant problems in this issue.

closed 2025-01-16 by wj-xiao