sipeed/NanoKVM #248: tcpdump and aircrack?

bionicrocky, 2025-01-07:
Why are tcpdump and aircrack-ng included on the base image?  They both have legitimate uses, but I fail to see what they are in this case.  What is the purpose?

Gunni, 2025-01-07:
tcpdump has been useful for me, Wifi on a kvm, is not though.

archont94, 2025-01-07:
> tcpdump has been useful for me, Wifi on a kvm, is not though.

NanoKVM have WiFi

bionicrocky, 2025-01-07:
The lite does not.  My question is why these would be included by default.  Aircrack is a wireless security audit/pentest tool. Why would/should this Devi e have it preloaded?

archont94, 2025-01-07:
> The lite does not. My question is why these would be included by default. Aircrack is a wireless security audit/pentest tool. Why would/should this Devi e have it preloaded?

Because they use same image for both versions.

Snazzie, 2025-01-08:
> > tcpdump has been useful for me, Wifi on a kvm, is not though.
> 
> NanoKVM have WiFi

my nanoKVM doesnt seem to be broadcasting a wifi ap, do i need to attach an antena?

archont94, 2025-01-08:
> i have the full version with screen and case.

Only NanoKVM PCIe have WiFi

bionicrocky, 2025-01-08:
Kind of getting away from the point of why Wi-Fi pen testing tools and packet capture tool are preloaded.  Why are they there?

archont94, 2025-01-08:
> Why are they there?

My guess is that they are left for debugging purposes - I was able to confirm Ethernet hardware problems with tcpdump (https://github.com/sipeed/NanoKVM/issues/223#issuecomment-2563577730), so they are useful IMO.

If you are that scared, you can always create own init script in /etc/init.d and call `chmod -x` on binaries of your choose (or delete them, but I'm not sure if this will persist boots), but that's seems to be quite paranoid. 

You can always build your own image, SDK is open source.

bionicrocky, 2025-01-16:
I’m not “scared of it”, it is just really odd that they are included by default as they have nothing to do with core functionality and they /can/ be dangerous in the wrong hands.  I don’t see any enterprise infosec group OKing the device with it.  I could find use for them for sure, but I’d expect to have to add them.

Gunni, 2025-01-16:
Disagree on tcpdump, I want to keep that one.

SANHAJIFX, 2025-05-13:
Any update guys on this thread? Does the latest version propose option to enable disable this (disabled by default)?

pkolega, 2025-05-18:
please add options in gui: remove aircrack-ng and tcpdump