sipeed/NanoKVM #249: [FEATURE REQUEST] Add Mouse Jiggler

lwbt, 2025-01-07:
> If user has enabled Mouse jiggler feature, It will allow to monitor the remotely connected device, by preventing it from getting into sleep mode or locking itself (plus would not require installation of additional software or purchase of USB jiggler)
>
> PiKVM has an implementation, and it would be a great addition to NanoKVM.
It will also allow me to buy more NanoKVMs thanks to not needing to buy a mouse jiggler too.

https://github.com/sipeed/NanoKVM/issues/158

I reopen this request as I have received my JetKVM an this is a feature that immediately jumps at me on the first screen after login. It's not a tucked away expert feature.

> **Jiggler**
> Simulate movement of a computer mouse. Prevents sleep mode, standby mode or the screensaver from activating

Snazzie, 2025-01-08:
Why cant you just disable sleep and auto lock in the OS

Snazzie, 2025-01-08:
Its literally a valid challenge to your idea. If you cant give a reasonable problem that is unsolvable unless this feature is implemented, then its not really a needed feature. 

As stated, your issue is directly solvable without this feature at OS level.

Its not patronizing or gaslighting to challenge ideas.

lwbt, 2025-01-08:
Check your privileges and have a little empathy man, not everyone can be a smug jerk working perfect products.

Sure, your much smarter than me, but that doesn't change the fact that the world outside your tiny bubble still looks like a dumpster fire.

Snazzie, 2025-01-08:
> Check your privileges

Ive seen enough. have a nice day.

ucki007, 2025-01-08:
I have to take @Snazzie 's side. 

The NanoKVM is a very low cost part of hardware with a limited amount of ressources. It is designed for remote controlling  / installing hardware and should run stable (each new component increases the chances of instabilities) and not suffer from something I like to call "featuritis" (an overload of features).

For me, even VPN (via tailscale) is one feature to much and to proof the stability aspect: we see in some other posts that it breaks the hardware limits of nanoKVM in some constellations / versions. But I get the point that this allows access to remote locations without configuring portforwarding etc.

Something like a mouse jiggler (or even debugging tools like aircrack and tcpdump) should not be part of a KVM as discussed in #248 . If something like that is needed it should be implemented on the host / network itself (e.g. disabling lockscreen / standby or installing the jiggler in Host OS).

dudududodododedede, 2025-01-09:
this would be a great feature for my work laptop -- I cannot install software on it, and can't disable locksreen/standby/screensaver

nmcbride, 2025-01-13:
yep agreed it would be a much appreciated feature

i looked at doing it with a script but they are not well documented and from what I could find it doesn't seem possible

should be a simple enough feature to implement it and if enough of their customers ask for it then it'd make sense for them to implement it. i have a feeling more people would want it and not use it than want to  use it and not have it.

monster1025, 2025-02-27:
+1 for jiggler functionality.
My company notebook has enabled 'hibernation mode' after two days of standby, and I can't disable it =(
Software mouse movements are ignored.

mhalano, 2025-05-02:
The changelog for version 2.2.6 mentions support for mouse jiggler. That's accurate?

civictd04, 2025-05-06:
> The changelog for version 2.2.6 mentions support for mouse jiggler. That's accurate?

Seems to work in 2.2.6 but someone reported you can't turn it off. I haven't tested turning it off.

mcbyte-it, 2025-05-08:
How does it work in 2.2.6? Where do I find the option to Jiggle the mouse?

civictd04, 2025-05-08:
[image]

closed 2025-05-12 by wj-xiao