sipeed/NanoKVM #250: Cant Acces Device after flashing SD Card

haniham, 2025-01-08:
Hello,
I today recieved my new NANO KVM. With the included SD Card i was somehow able to set up a static IP. 
The DEvice has no Internet Access and never will have.
I decided to update the Device to 1.3.0 via balena ether afther this it complained on the oled that it cant access the network or something like this
I flashed back to 1.2.1 and now it complains on the OLED "Couldn't get IP"
In both cases i have set a eth.nodhcp with content "192.168.10.20/24"
What is going on - dos this require internet Access?

gregoreau, 2025-01-14:
In my experience, yes it requires internet access for initial setup to download and flash to the latest application version (firmware) which is not the same as the image version that you flash to the UI.
Cross-post to another issue which what worked for me:
https://github.com/sipeed/NanoKVM/issues/186#issuecomment-2589838447

haniham, 2025-01-16:
I Tried the following: 

Flashed 1.3.0
Ethernet and Power plugged in with no internet Access

Device hangs in a Loop  (The Adress was served by my test DHCP)
```
Auto-Updating
192.168.177.50
``` 
after some time:
```
Please Connect
to the network
to update app
``` 
It has network, but no Internet
Then it resets and goes back to the updating

I needed to go home so i have Internet access 😒
If i connect my testnet to the internet 
```
Auto-Updating
192.168.177.50
``` 
after some time it boots to the usual Screen with IP and data displayed

So i made a dd dump of the completed Image - was 32GB and zipped it to only 306MB, so just i slight increase from the original Image with 179 MB - why cant we get a complete image. These hardware is not supposed to see any Internet

joshuagamboa, 2025-02-04:
any idea if this can be done a Mac machine? kpartx does not exist in the Mac terminal.

wj-xiao, 2025-02-19:
Manual updates are now supported, as mentioned by @gshpychka .

closed 2025-02-19 by wj-xiao