sipeed/NanoKVM #251: Upload Images via web interface

jcv8-dev, 2025-01-08:
It would be nice if we could upload iso images via the web interface instead of scp. This would greatly streamline the user experience. From a technical perspective this should be possible if the cpu can handle the file transfer. At least the PiKVM is able to handle the file upload via HTTP.

stuartm, 2025-01-30:
It might be obvious to many, but there are other files, aside from images which it would be useful to upload to storage via the UI - BIOS Firmwares for one. Similarly being able to download files back from storage i.e. BIOS configuration backups.

Other times it would just be helpful to have a quick method to transfer small files such as config files via the NanoKVM UI for situations where the host machine is booting to an emergency shell without networking or secure shell. The real beauty of a KVM like NanoKVM is that I can troubleshoot issues from anywhere, even if I just have a phone/tablet without an SSH client installed.

dormancygrace, 2026-09-14:
Hi! This feature is available in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

The Web UI includes ISO upload from your computer to the NanoKVM SD card. Select a local ISO in the Download panel; it is transferred over HTTP and saved in /data, so SCP is not required. The upload path validates the ISO and supports an optional SHA-256 checksum.

This covers the original ISO-upload request. The broader request in the comments for a general-purpose file manager, arbitrary BIOS/configuration files and downloads back to the browser is separate. No upload-speed claim is being made.

[Web UI implementation](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/menu/download.tsx#L266) · [Server upload handler](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/service/download/service.go#L204).

If you still need this feature, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!