sipeed/NanoKVM #253: How to start a script at startup?

MatejKovacic, 2025-01-09:
I am trying to run Wireguard VPN to my server.

I created /etc/wireguard/wg0.conf with:
```
[Interface]
PrivateKey = ****

[Peer]
PublicKey = ****
PresharedKey = ****
Endpoint = xx.xx.xx.xx:51194
AllowedIPs = x.x.x.0/24
PersistentKeepalive = 5
```

Then I run:
```
ip link add dev wg0 type wireguard
wg setconf wg0 /etc/wireguard/wg0.conf
ip addr add x.x.x.x/32 dev wg0
ip link set up dev wg0
ip route add x.x.x.0/24 dev wg0
```

This works fine. But how can I start my VPN automatically at startup (reboot)?

There is no system, `crontab`, if I put these commands to `/etc/rc.local` they do not run, also if I create `/etc/init.d/S99wireguard` this doesn't run.

Any help?

MatejKovacic, 2025-01-09:
OK, really ugly hack is to add these commands at the end of `/etc/init.d/rcS`. But I would prefer some more clean solution...

stvorl, 2025-01-10:
You can add custom "service" to /etc/init.d with proper priority prefix (10, 99, etc), just base it on any script here, and re-implement start) and stop) cases. Like I responded with in https://github.com/sipeed/NanoKVM/issues/245.
At least, that would be piece of logic, concluded in separate file, not changing OEM files.
But, maybe this solution is uglier than your, indeed. 😉

MatejKovacic, 2025-01-10:
Okay... I created `/etc/init.d/S50myvpn`:

```
#!/bin/sh
case "$1" in
  start)
        echo "Start my Wireguard connection"
        ip link add dev wg0 type wireguard
        wg setconf wg0 /etc/wireguard/wg0.conf
        ip addr add xx.xx.xx.xx/32 dev wg0
        ip link set up dev wg0
        ip route add xx.xx.xx.xx/24 dev wg0
        ;;
  stop)
        ;;
  restart|reload)
        "$0" stop
        "$0" start
        ;;
  *)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
esac
exit $?
```

... and is not working. I. e. does **not** autostart. Do I need to run anything else? If commands are in `/etc/init.d/rcS`, it works.

MatejKovacic, 2025-01-10:
Sorry, never mind. I forgot to make it executable (`chmod +x /etc/init.d/S50myvpn`).

closed 2025-01-10 by MatejKovacic