sipeed/NanoKVM #255: HID device stops working sometimes

MatejKovacic, 2025-01-10:
Quite often HID device (mouse, keyboard) stops working. Sometimes it helps to click "Reset HID device", but sometimes I need to ssh to NanoKVM and reboot it...

What are the solutions?

DaisukeMinagawa, 2025-01-10:
I have the same issue.

MatejKovacic, 2025-01-10:
Found something interesting. It seems that it happens where the target computer (computer connected to NanoKVM) starts screensaver (i. e. lock current session).
Could be some change of a resolution or something?

DaisukeMinagawa, 2025-01-10:
@MatejKovacic  It is not caused by the screen saver or locking the current session. I encountered the problem while using grab shell of Linux.
I'll try to display mode to MJPEG.

svedsvedsveden, 2025-01-10:
I gave the same issue - mine do not seem to be connected to screensaver. It is completely random, and quite often.

Do any of you also have problems with tailscale staying connected, and just working in general?

MatejKovacic, 2025-01-10:
Correct, found out that session got locked, abut HID worked.

I am using my own Wireguard VPN. I found out that sometimes connection drops a little, I also go disconnected from ssh session, but in general no bigger problems.

DaisukeMinagawa, 2025-01-11:
@MatejKovacic I found HID does not work after mounting .iso image.

stvorl, 2025-01-11:
I found, than in fully working high-level OS (Windows, Linux, etc) mounting and unmounting ISO working well, HID not ceased.

This issue happens only in low-level apps (UEFI/BIOS setup, GRUB loader console and so on), and, looks like, may not reproduce on some motherboards.

Anyway, you can still mount ISO image, then reboot controlled PC, re-enter low-level app, and now HID is working again, while image is mounted too.

Looks like, when we're mounting disk, unit is re-connecting its "USB-hub-with-keyboard-and-mouse-and-disk" emulation device (to force controlled PC to re-detect devices on it, because there is no another way), and there is some troubles in device introduction logic on unit, interfered with low-level HID device detection on PC.

MatejKovacic, 2025-01-11:
Now it seems to me that it is more or less random behavior. With no clear patterns. But happens quite often.

stvorl, 2025-01-11:
> Now it seems to me that it is more or less random behavior. With no clear patterns. But happens quite often.
What OS is under control while it happens?
Are you sure it is HID issue? Can it be just video stream interruption due to poor communication? 
I found some cases in https://github.com/sipeed/NanoKVM/issues/257. Does it help to reopen WebUI in browser?

MatejKovacic, 2025-01-11:
OS under control is Ubuntu. It was not restarted, connection was not lost, because I can ssh to NanoKVM.

MatejKovacic, 2025-01-11:
Sometimes helps to reopen webui, sometimes "reset HID" helps, but more often I need to reboot NanoKVM.

stvorl, 2025-01-11:
I had exactly same looking situation, but real issue was in video streaming freeze. 
I found that sometimes, when connection is poor (like ping losses and so on), video seems to be suspended forever (till WebUI reopening) but HID commands was still delivering to controlled device.

And I can still SSH to KVM (because connectivity returns quickly), and even existing SSH connection not breaks (because I'm not using it right in time when I was busy by punching frozen WebUI, when connection was shortly in hang)

Reopening WebUI in another tab will restore video streaming and I can read text I blind typed on terminal recently. Did you try to blind write some text in terminal before WebUI reload, and then check for it after?

Reopening WebUI can also point to some frontend and browser issue, or OS related browser issue. Sure, when you reboot the unit, it will cause reopening WebUI too. 

Maybe to test with another browser, non-webkit? Like, Firefox (gecko)?

mcury1985, 2025-01-12:
Same problem here.
One thing that I've been doing to mitigate the issue is disconnect the keyboard from the host.
By disconnecting the USB keyboard from the host, I don't need to reboot or reset HID anymore.

MatejKovacic, 2025-01-12:
> Maybe to test with another browser, non-webkit? Like, Firefox (gecko)?

I tested Chromium and Firefox. When HID stops working, it is the same.

stvorl, 2025-01-13:
@mcury1985, @MatejKovacic 
This is really bad. I didn't test the unit for a long time work, just for hour-two, on about few motherboards and RPi. Maybe this is reproducing with mine too, maybe not, I'll try tightly next week. Maybe this issue is reproducing only on crossing with specific motherboards.

If your controlled PC is linux-based, it need to be checked with **dmesg** and **lsusb** commands on next such event (i mean, HID disconnect), using directly connected keyboard, in order to observe USB bus errors and USB device "Sipeed gadget" presence. 

I think, this would be useful for maintainers, and musthave for their response. If they'll come down to read and handle this, 'course. 🤭

mcury1985, 2025-01-13:
> If your controlled PC is linux-based,

Its both Windows and Linux (EndeavourOS which is based on Arch Linux).
Controlled PC is dual boot.
The problem happens to switch the OS pressing F8 to show boot options or DEL to get into the BIOS when booting from OS to the other.
The only way it always work is by unplugging the USB keyboard from the host (controlled PC).

Perhaps not related to this problem, not sure, but a little explanation about my system:
You can ask me, well, why don't you use GRUB.. I don't like GRUB, I use two hard drives, one for each OS, both of them have their own EFI partition.. This avoid the Windows update problem messing with my Linux.
Linux is systemd-boot btw.

wj-xiao, 2025-01-16:
We've fixed the “Reset HID” in the version 2.1.5.

Please update to the latest version and try resetting the HID if it's not working.

snova69, 2025-01-22:
updated to 2.1.5 problem still happening

snoopsbne, 2025-01-22:
Same 2.1.5. Reset HID works sometimes but most of the time a reboot of the NanoKVM is needed. The issue is present before any OS loads (so Bios/UEFI level) on the host. Issue occours more often when using virtual images.

snova69, 2025-01-23:
I think these issues are related to Intel MEBx/AMT being left on in the BIOS. It does trickery to present a HID as well to the BIOS and OS.  I turned my MEBx/AMT off and things seem to be better.  While you are at it disable the internal Ethernet which MEBx/AMT uses for remote management.

snova69, 2025-01-23:
so I went into the BIOS enabled the MEBx hotkey. In my case it is Control-P on a Dell.  Exited the BiOS..  it rebooted and I hit Control-P to enter MEBx/AMT.   Entered the MEBx/AMT password.  Then proceeded to disabled the Management Engine.   Saved and then exit.  Again reboot the machine and back into the BIOS.   Turn off the MEBx hotkey and enabled the internal NIC.   So far so good.   Dont need to reboot the NanoKVM due to Keyboard/Mouse HMI interface.  Pretty sure it my it it was a fight between the NanoKVM presentation of Keyboard/Mouse over USB vs Intel MEBx/AMT presentation of mouse/keyboard to BIOS and OS using trickery of their own.

stvorl, 2025-01-24:
> The issue is present before any OS loads (so Bios/UEFI level) on the host

@snoopsbne 
Creating /boot/BIOS semaphore file, whatever that means, helped me to fix HID in low-level apps (BIOS, EFI Shell, GRUB).
Please, try:
# touch /boot/BIOS
# reboot

Z2Z-GuGu, 2025-03-26:
In most motherboards, the BIOS requirements for HID devices are not very strict, and the normal mode can be recognized by the BIOS. However, some motherboards require the creation of /boot/BIOS in NanoKVM to activate a special mode for the keyboard and mouse, which has helped many users.

That said, some motherboards, such as the HP Elite series, do not benefit from creating /boot/BIOS, as their BIOS may disable HID devices when combined with other devices. To address this, we introduced 'HID Only' mode in version 2.2.5, which completely eliminates the IAD descriptor in the USB to improve compatibility with such BIOS. This feature can be enabled by clicking 'HID Only' in the mouse icon.

closed 2025-03-26 by Z2Z-GuGu