sipeed/NanoKVM #257: WebUI user-frendly display improvements on "no signal" and communication issues.

stvorl, 2025-01-11:
While the controlled device is shut down, the WebUI will display the last frame seen, and do it until the signal is reconnected or the browser is reopened. 

Similarly, if there is a commlink delay or interruption between the frontend and backend due to poor internet quality, the last frame may be displayed for a long time, not showing any hint about connection troubles. And when connections returns, and even HID returns to works, video sometimes still frozen until WebUI re-opened.

So, being located miles away from controlled device, working on an operating system reinstallation or other tasks that require multiple reboots, and knowing, than NanoKVM unit still buggy, I can't really distinguish what is happening, when image freezes:
- Is the controlled device shut down?
- Is the frozen?
- Has the NanoKVM unit frozen it's HID susbystem due to bugs, causing the controlled device to not receive inputs?
- Has the unit frozen it's HDMI module?
- Has the frontend lost video connection to the backend?

So, can you please consider to make some improvements to KVM module?
- Add "No signal" notification, or make screen just blank, when it's really no signal.
- Make frontend do some heartbeet requests to backend, showing "Communication ceased" notification when it is temporarily no response.
It may be also helpful to pause HID events transmission during periods of temporary loss of communication, to avoid situations, in which user is pressing hysterically on all buttons (and they're delivered to device, making "chaos and devastations" on it) while video transmission is frozen.

AJolly, 2025-01-13:
>   It may be also helpful to pause HID events transmission during periods of temporary loss of communication, to avoid situations, in which user is pressing hysterically on all buttons (and they're delivered to device, making "chaos and devastations" on it) while video transmission is frozen.

This seems like a very bad idea, considering you often need input to wake up the video.

wj-xiao, 2025-01-16:
Currently, the program simply receives data from HDMI and transmits it to the web interface. 
There could be many reasons why the web has no display, but the program doesn't differentiate between these errors and report to the user. 

We will improve this process by adding more status indicators to provide better feedback.

archont94, 2025-01-28:
@wj-xiao there is already [verification of HDMI link state here](https://github.com/sipeed/NanoKVM/blob/97a9b376c79611ca1cceb8f6df282670bb597379/support/kvm_system/main/lib/oled_ui/oled_ui.cpp#L268C6-L268C25), so why cannot it be used to show dummy `no signal` message? I'm not sure how `go` backend server communicate with `kvm_system`, but this information is already there.

What is even the reason to split stuff into C++ `kvm_system` and go `server`? Wouldn't it be easier to have enerything in one place?

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The viewer includes a capture-status overlay for missing HDMI input and capture errors. The USB/control toolbar icon also shows reconnection in amber and repeated connection failures in red, and WebSocket heartbeats detect silent control-connection loss.

These cover the signal and connection feedback requested here. They do not reliably distinguish every host freeze, HDMI-driver hang or physical USB failure, and do not establish that all frozen-video recovery cases are fixed.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/capture-status/overlay.tsx) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/menu/usb.tsx).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!