sipeed/NanoKVM #258: NanoKVM Terminal not loading

matouka, 2025-01-11:
Yesterday I noticed that there was an update to install on my NanoKVM Lite, so I updated. My image version is v1.3.0 and the application version is v1.3.0.

The problem is, when I click on NanoKVM Terminal now, which used to work, I now get the screen below and can't seem to log in.

[image]

wj-xiao, 2025-01-13:
In version 2.1.4, we've updated the terminal login logic.
For security purposes, changing password via the web page now also updates the system's root user password. 

NanoKVM has two user accounts:
1. a web user for accessing the web interface
2. a system root user for SSH logins

Previously, changing the password only affected the web user. But in the latest version, it updates both the web user and the root user.

Consequently, the web terminal will now use this newly updated password for login. This introduces an issue: if you changed your password in a prior version, that password wasn't applied to the root user. 

You can resolve this in two ways:
1. Change your password through the web page again. 
5. Log in manually using the default password 'root'.

matouka, 2025-01-13:
Thanks. Worked perfectly!! This update seems to be considerably more performant. Well done!