sipeed/NanoKVM #260: TypeError: Failed to fetch dynamically imported module: ***/assets/index-uEuGYS6A.js

sizkein, 2025-01-15:
[image]

The web page is not stable.

Sometimes when I access it, I only see a white screen,

or I get a module error like the screenshot.

If I press Ctrl+F5 several times, it works fine.

What could be the problem?

wj-xiao, 2025-01-16:
This is due to browser caching. 

It often happens after an application update, where the browser is still using cached files from the old version.
Force-refreshing(Ctrl+F5) the browser, as you did, generally fixes this.

sizkein, 2025-01-16:
When I press Ctrl+F5, it doesn't always work properly.
I have to repeat it 3~4 times to get it to work properly.
It's been like this from the beginning, and I thought it was a system problem, so I installed versions 1.2, 1.3, and the latest version, but the same problem occurs.
It doesn't happen every time, but when I access the page, it doesn't work properly the first time.

Gitllub, 2025-09-24:
@sizkein @wj-xiao  It's happened to me too, Did you find any solutions ?