sipeed/NanoKVM #261: Feature Request - Screenshot function

rcarmo, 2025-01-15:
Hi! It would be great if we could take screenshots of a device and download them in PNG or JPEG format

clipcarl, 2025-01-20:
Is this really needed when the browser can already save images of what's displayed? For example in Firefox I can shift-right-click on the NanoKVM window and select "Take Snapshot" or "Take Screenshot" to get a screenshot of what's displayed.

rcarmo, 2025-02-02:
Yes, of course it is needed. Not everyone uses Firefox (in fact, sadly, less and less people do), and if you are using the KVM to write documentation then you are likely to take multiple screenshots during a session (many KVMs also have a recording function). This, by the way, is a feature that has been in the first PiKVM for years.

clipcarl, 2025-02-02:
Obviously, it's not just Firefox that can do page screenshots. You can do them in Chrome through the developer tools and there are multiple extensions that make it easy. Same with every other major browser.

Further, every major OS has a way to screenshot windows or rectangular areas which are available at the tap of a single keypress. So I'm not sure why the NanoKVM developers need to reinvent the wheel for a feature that's already incredibly easy to accomplish two other ways.

rcarmo, 2025-02-02:
I don't think we have the same use case scenarios (or control) for where and when these things are used, or with which clients. You assume people have the flexibility or skills to do that through other means without (apparently) being anywhere near involved with the project.

Suffice it to say that a) this is a feature that is relied upon for documentation b) it is built in to other KVMs c) it is not rocket science to implement.

clipcarl, 2025-02-02:
> Suffice it to say that a) this is a feature that is relied upon for documentation

Yes, I've used it extensively for that purpose myself. And the OS screenshot tool works perfectly for that.

> b) it is built in to other KVMs

I'm not sure this is true. I don't remember seeing this option in any of the web browser based IPMI (or external) KVMs I've worked with. I think the web browser based ones just assume you can use the browser or OS for screenshots.

> c) it is not rocket science to implement.

Yeah, it's easy to say that when it's not **your** time and money. If you really meant it you'd develop the feature yourself and send in a patch.

clipcarl, 2025-02-02:
[image]

One tap of a key.

rcarmo, 2025-03-06:
In case other people come across this and are equally appalled at the "help" from random strangers but really need to capture _exactly_ what the video stream is rendering and nothing more, a tenable workaround do do that and automatically download a screenshot is to create a browser bookmarklet that copies the current frame to a canvas and downloads that:

```javascript
javascript:"use strict";!function(window,document){var canvas=document.createElement("canvas"),video=document.querySelector("video"),ctx=canvas.getContext("2d");canvas.width=parseInt(video.offsetWidth);canvas.height=parseInt(video.offsetHeight);ctx.drawImage(video,0,0,canvas.width,canvas.height);var currentDateTime=new Date().toISOString().replace('T',' at ').replaceAll(':','.').substring(0,22);var a=document.createElement("a");a.download=`Screenshot ${canvas.width}x${canvas.height} ${currentDateTime}.jpg`;a.href=canvas.toDataURL("image/jpeg");document.body.appendChild(a).click();a.remove()}(window,document)
```

(adjust filename to taste)

This can be added to the web UI - if I manage to set up a build pipeline, I'll submit a pull request.

clipcarl, 2025-03-06:
Your new solution is exactly what I already told you: use the browser. Glad you finally realized that's the way to go.

dormancygrace, 2026-09-14:
Hi! This functionality is available in [NanoKVM OS beta-10](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v1.0.0-beta.10), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

A camera button in the viewer downloads the current video frame as a PNG, with a timestamp in the filename. It supports Direct, WebRTC and MJPEG and excludes the Web UI toolbar.

The implemented download format is PNG. The image uses the decoded stream dimensions, which may be smaller than the HDMI input when downscaling is enabled.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/web/src/pages/desktop/menu/screenshot.tsx) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/web/src/lib/screenshot.ts).

You're welcome to try it and let me know how it works on your setup. Development testing has been on PCIe/UXC; Cube qualification is still pending. Installing from stock requires flashing the full SD image, so please back up your card first. Existing beta-9 installations with the compatible 64 MiB boot layout can use the signed system update.

[Release and downloads](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v1.0.0-beta.10) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/12a1e169727eab73dd335049eb9d67fc427e2035/docs/INSTALL.md)

Please include your board revision, HDMI source, browser, codec and transport when reporting results. Thanks!