sipeed/NanoKVM #264: PCIe black bracket misaligned

ancan, 2025-01-16:
I don't know if it's the right forum to address hardware issues, but I just received my card which has the black bracket, and it won't fit in the computer. I've compared to other PCIe cards and the bracket is about 5mm off so even with the card bottomed out in the slot you can't get the part you screw it to the case to be flush. Bad batch or all black brackets?

Zepan, 2025-01-17:
Hi, a batch of black brackets was specially customized for specific cases, which are 3mm longer than those for standard cases. Some of these mistakenly got mixed into regular orders during packagement.
As the picture shows, it is still able to install to the normal case, as the screws are 5~6mm long, just take care to not screw to much force.
[image]

If you want a replacement, you can contact custom service, for a direct replacement; or send together with your new order(and we will give you $5 coupon for new order)

ancan, 2025-01-17:
Thanks. I'll contact customer service and ask for just a new bracket.

wm-ytakano, 2025-01-20:
I’ve encountered the same issue with the black bracket. Here are the details of my experience:
- The silver bracket from a previous order fit perfectly, but the black bracket from my recent order does not align properly.
- My setup:
  - Motherboard: GIGABYTE X870E AORUS PRO ICE
  - Case: Lian Li LANCOOL 207
- The screws provided were too short to secure the card. As a temporary fix, I used longer screws intended for HDD mounting. This workaround helped, but it's not ideal for long-term use.

It would be helpful if the official website provided a clear explanation regarding this batch of black brackets.

[image]

[image]

kungfoome, 2025-01-26:
I have the same problem. I dont use screws in the cases where I want to install this in, so I can't use it without the correct bracket.

hesslr, 2025-02-03:
@Zepan – I sent a request for a replacement bracket to kvm@sipeed.com as suggested, then followed up with support@sipeed.com a day later to ensure it reached the right team. However, I haven’t received any response in several days.

Could you clarify the correct support process for requesting the correct bracket? Should I be contacting a different email or using another method?

Thanks for your help!